libfive/libfive #291: Bridging a gap between shapes.

lshoravi, 2019-07-16:
Hi again, and thanks for the great program. Having lots of fun with it!

I'm writing here to ask for help with a perticular problem that I haven't found a good solution for- I hope that's ok. I couldn't find an IRC or similiar to ask questions at, so I figured an issue would do.

I'm looking for a "clean" (read: relatively concise and algorithmic) way to bridge gaps between shapes. It looks like this:

[image]

Pictured is a couple of curved columns with different height offsets, and what I want to do is "fill in" the gaps in a satisfactory way. I also screenshot a majority of the code- I hope it's legible.

Thanks!

lshoravi, 2019-07-16:
I'll also paste it for convenient access, so you don't have to write if you want to test:

```
(set-bounds! [-150 -150 -150] [150 150 150])
(set-quality! 1)
(set-resolution! 1)

(define (deg->rad deg)
  (* deg (/ pi 180)))

;;; Values
(define pi 3.1415926535)

 ;; Frame measurements TODO: Document where these come from.
(define inner-width 14.0)
(define padding 2.5)
(define plate-height 2)
(define full-width (+ inner-width (* padding 2)))

 ;; Angle measurements.
(define row-angle (deg->rad 5))
(define col-angle (/ (* (deg->rad 14) 2) 6)) ;; Assumes 6 columns
(define row-point [0 0 220])
(define col-point [0 0 240])

 ;; Offset measurements.
(define offsets #@1(0 0 -3 -5 0 0)) ;; Assumes 6 rows
(define extra-rotations #@1(0.4 0.2 0 -0.2 0 0))

;;; Helper functions
(define (plate-aux row col struc)
  (let assemble ((r row)
                 (c col)
                 (acc '()))
    (cond ((zero? r) acc)
          ((zero? c) (assemble (1- r) col
                               (cons (place r col struc) acc)))
          (else (assemble r (1- c) (cons (place r c struc) acc))))))

;;; Structures
(define inner-frame (box-mitred-centered [inner-width inner-width plate-height]))
(define outer-frame (box-mitred-centered [full-width full-width plate-height]))
(define frame (difference outer-frame inner-frame))

;;; Placement functions
(define (place row col struc)
  (sequence struc
            (move [0 0 (array-ref offsets col)])
            (rotate-x (* (- row 3 (array-ref extra-rotations col))
                         row-angle)
                      row-point)
            (rotate-y (* (1- col) col-angle) col-point)))

(define (plate row col struc)
  (apply union (plate-aux row col struc)))

;; Playground
(plate 5 6 frame)
```

mkeeter, 2019-07-17:
That's an interesting question, and a cool model – are you building a mechanical keyboard?

My first instinct is to use the `blend` function, which does blending between distance fields.

It's not perfect, but by tweaking the blend parameter, you can at least join up the different rectangles.  I've changed the design slightly to punch the centers out afterwards, so we're building a blended base then subtracting out the inner frames.

[image]

The downside to this approach is that it creates bumps where two fields are blended and summed.  You can somewhat mitigate this by changing the blend parameter, but it's a balancing act – if it's too small, the fields won't meet, and if it's too large, they'll be bumpy.

Here's the changed version of the code:

<details><summary>Click to expand</summary>
<p>

```
(set-bounds! [-150 -150 -150] [150 150 150])
(set-quality! 1)
(set-resolution! 1)

(define (deg->rad deg)
  (* deg (/ pi 180)))

;;; Values
(define pi 3.1415926535)

 ;; Frame measurements TODO: Document where these come from.
(define inner-width 14.0)
(define padding 2.5)
(define plate-height 2)
(define full-width (+ inner-width (* padding 2)))

 ;; Angle measurements.
(define row-angle (deg->rad 5))
(define col-angle (/ (* (deg->rad 14) 2) 6)) ;; Assumes 6 columns
(define row-point [0 0 220])
(define col-point [0 0 240])

 ;; Offset measurements.
(define offsets #@1(0 0 -3 -5 0 0)) ;; Assumes 6 rows
(define extra-rotations #@1(0.4 0.2 0 -0.2 0 0))

;;; Helper functions
(define (plate-aux row col struc)
  (let assemble ((r row)
                 (c col)
                 (acc '()))
    (cond ((zero? r) acc)
          ((zero? c) (assemble (1- r) col
                               (cons (place r col struc) acc)))
          (else (assemble r (1- c) (cons (place r c struc) acc))))))

;;; Structures
(define inner-frame
  (rectangle-centered-exact
    [inner-width inner-width]))
  
(define outer-frame (box-mitred-centered
[full-width full-width plate-height]))
(define frame (difference outer-frame inner-frame))

;;; Placement functions
(define (place row col struc)
  (sequence struc
            (move [0 0 (array-ref offsets col)])
            (rotate-x (* (- row 3 (array-ref extra-rotations col))
                         row-angle)
                      row-point)
            (rotate-y (* (1- col) col-angle) col-point)))

(define (plate row col struc)
  (fold (lambda (a b) (blend a b 2)) (emptiness)
        (plate-aux row col struc)))

;; Playground
(difference
  (plate 5 6 outer-frame)
  (apply union (plate-aux 5 6 (scale-z inner-frame 5))))
```

</p>
</details>

A more robust approach would be to define your curved substrate beforehand (generating the entire surface, rather than individual frames), then punch out the frame holes using the same `plate` logic as before.

lshoravi, 2019-07-17:
It is indeed a mechanical keyboard.

The blend technique looks good but as you say the bumps prove to be a problem- it is likely that the switches won't be able to sit nicely on the plate. I could however try taking that with the difference of a raised unblended plate. It looks like this:

[image]

changed code snippet being:

```scheme
(difference
  (plate 5 6 outer-frame)
  (apply union (plate-aux 5 6 (scale-z inner-frame 5)))
  (apply union (plate-aux 5 6 (move (box-mitred-centered [full-width
                                                          (+ full-width 5)
                                                          plate-height])
                                    [0 0 plate-height]))))
```

lshoravi, 2019-07-17:
Actually, filling the gaps on one direction was trivial: 

```scheme
(define top-filler (box-exact [(/ full-width -2) (/ full-width 2.15) (/ plate-height -2)]
                              [(/ full-width 2) (/ full-width 1.85) (/ plate-height 2)]))
(plate 4 6 top-filler)
```