libfive/libfive #297: How to build Studio on Linux/Ubuntu?

rem3-1415926, 2019-08-05:
### Description

I am sorry for this probably pretty stupid noob topic. But as a total Linux newbie who just set up an Ubuntu VM and has hardly any _practical_ programming experience, I am totally lost. I couldn't find any kind of information about what I am supposed to do with that code I cloned to my disk.

I looked around and managed to execute the following commands in the terminal without any errors, and seemingly successfully (in this very order): 

`sudo apt install git` 

`git clone https://github.com/libfive/libfive` 

`sudo apt-get install cmake pkg-config libeigen3-dev libpng-dev libboost-all-dev qtbase5-dev guile-2.2-dev` 

`cd libfive` 

`mkdir build` 

`cd build` 

`cmake -DCMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/qt5/bin ..` 

`make all` 

`cmake -DGUILE_INCLUDE_DIR=/usr/include/guile/2.2/ -DGUILE_LIBRARY=/usr/local/lib/libguile-2.2.so -DCMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/qt5/ ..` 

(I have no idea what half of these might possibly be doing, in case i screwed up, but you gotta start somewhere, right?) 

What am I missing? What is there left to do to get the Studio somehow executable? Or am I just retarded and it already is there but I can't find it? 

And as I'd rather learn something than just stay as oblivious as I am in this topic right now, I'd very much appreciate short descriptions of what the commands are actually doing, if someone's bored enough and doesn't mind helping me out with that. (I do get the linux shell ones in their basic form and "make all")

Thanks very much in advance.

### Versions

- Operating system: Ubuntu 18.04.2 LTS

NoNeedToHurry, 2019-08-29:
Don't know if it's too late but I had the same trouble on elementary OS 5.0 Juno (based on Ubuntu 18.04).

I followed the same instructions and found weird that despite guile-2.2-dev was installed, cmake says :

> -- Checking for module 'guile-2.2>=2.2.1'
> --   Found guile-2.2, version 2.2.3

And then can't find dependencies :

>   libfive-guile:	✘   (needs Guile 2.2 or later)
>   Studio:		✘   (needs Guile 2.2 or later)

So, I trusted package installation rather than cmake messages and tried a really dirty hack in the root libfive folder on CMakeLists.txt file by adding `SET(GUILE_FOUND TRUE)` in between two lines like that :
```
message("  libfive:             ✓")

SET(GUILE_FOUND TRUE)

if (BUILD_GUILE_BINDINGS)
```
I guess it's not the proper way but that did the trick for me because the studio was then compiled successfully.

Good Luck

mkeeter, 2019-08-30:
@NoNeedToHurry that was a mistake introduced in PR #301, which I've just now reverted (in 119f9b6c1d9cc7792860da2b7ea35d64f2358213): the PR added checking for Guile 3, but broke cases where only Guile 2 was installed.

@rem3-1415926 , can you please paste the entire console log from your attempt to build Studio?

rem3-1415926, 2019-08-30:
Thanks for the support. 
Here's the log, it does contain some pretty... moronic mistakes that made commands abort right away, so I cleaned them away in my original request to hide my shame. As I said, all of it executed seemingly successfully in the end.

`git clone https://github.com/libfive/libfive
sudo apt install git
git clone https://github.com/libfive/libfive
sudo apt-get install cmake pkg-config libeigen3-dev libpng-dev libboost-all-dev qtbase5-dev guile-2.2-dev
build\libfiv.sln
cd libfive
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/qt5/bin
cmake -DCMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/qt5/bin ..
make help
make all
cmake -DGUILE_INCLUDE_DIR=/usr/include/guile/2.2/ -DGUILE_LIBRARY=/usr/local/lib/libguile-2.2.so -DCMAKE_PREFIX_PATH=/usr/lib/x86_64-linux-gnu/qt5/ ..`

mkeeter, 2019-08-30:
That log only contains the commands that *you* ran – you also need to show the outputs (what the terminal prints back at out) for it to be used for debugging.