libfive/libfive #302: Building Studio from master fails on Arch Linux

yawor, 2019-08-29:
### Description

The build process fails on linking the Studio binary. Everything else (libfive and libfive-guile) seems to build properly. Here's an error output (using Ninja):

```[95/97] Linking CXX executable studio/Studio
FAILED: studio/Studio 
: && /usr/bin/c++  -Wall -Wextra -g -fPIC -pedantic -Werror=switch -march=native -lstdc++ -O3 -DNDEBUG -O3 -DRELEASE -DEIGEN_NO_DEBUG  -latomic -rdynamic studio/CMakeFiles/Studio.dir/Studio_autogen/mocs_compilation.cpp.o studio/CMakeFiles/Studio.dir/src/main.cpp.o studio/CMakeFiles/Studio.dir/src/args.cpp.o studio/CMakeFiles/Studio.dir/src/app.cpp.o studio/CMakeFiles/Studio.dir/src/arrow.cpp.o studio/CMakeFiles/Studio.dir/src/axes.cpp.o studio/CMakeFiles/Studio.dir/src/background.cpp.o studio/CMakeFiles/Studio.dir/src/bbox.cpp.o studio/CMakeFiles/Studio.dir/src/busy.cpp.o studio/CMakeFiles/Studio.dir/src/camera.cpp.o studio/CMakeFiles/Studio.dir/src/documentation.cpp.o studio/CMakeFiles/Studio.dir/src/script.cpp.o studio/CMakeFiles/Studio.dir/src/settings.cpp.o studio/CMakeFiles/Studio.dir/src/shader.cpp.o studio/CMakeFiles/Studio.dir/src/syntax.cpp.o studio/CMakeFiles/Studio.dir/src/shape.cpp.o studio/CMakeFiles/Studio.dir/src/color.cpp.o studio/CMakeFiles/Studio.dir/src/editor.cpp.o studio/CMakeFiles/Studio.dir/src/interpreter.cpp.o studio/CMakeFiles/Studio.dir/src/view.cpp.o studio/CMakeFiles/Studio.dir/src/window.cpp.o studio/CMakeFiles/Studio.dir/Studio_autogen/EWIEGA46WW/qrc_resources.cpp.o  -o studio/Studio  -Wl,-rpath,/home/yawor/src/libfive/build/libfive/bind:/home/yawor/src/libfive/build/libfive/src: libfive/bind/libfive-guile.so /usr/lib/libQt5OpenGL.so.5.13.0 /usr/lib/libQt5Concurrent.so.5.13.0 libfive/src/libfive.so /usr/lib/libpng.so /usr/lib/libz.so -lpthread -lguile-3.0 -lgc /usr/lib/libQt5Widgets.so.5.13.0 /usr/lib/libQt5Gui.so.5.13.0 /usr/lib/libQt5Core.so.5.13.0 && :
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<2u>::pushIndexTable(libfive::CornerIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<3u>::qefSumTable(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<2u>::cornerTable(libfive::CornerIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<3u>::neighborTable(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::EdgeTables<2u>::boundary(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::EdgeTables<3u>::subspaces(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<3u>::cornerTable(libfive::CornerIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<3u>::pushIndexTable(libfive::CornerIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::EdgeTables<2u>::subspaces(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<2u>::neighborTable(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::EdgeTables<3u>::boundary(libfive::NeighborIndex)'
/usr/bin/ld: libfive/src/libfive.so: undefined reference to `libfive::NeighborTables<2u>::qefSumTable(libfive::NeighborIndex)'
collect2: error: ld returned 1 exit status
```

It doesn't matter if I use make or ninja. It always fails on this step. This happens on current master, but the same happened before the last merge, which switched dependency to guile-3.

Please let me know if anything else is needed (software versions etc).

### Versions

- Operating system: Arch Linux
- `libfive` commit hash: `aa8579eb67761f0e395d5d5cdafa7a1b0177474f`
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.13

mkeeter, 2019-08-30:
I don't have an Arch Linux machine or image at hand, so you'll have to devise a fix and open a PR.

This looks related to explicit template instantiation.  One thing to check: do the `NeighborTables` error messages go away if you add

```
extern template class NeighborTables<2>;
extern template class NeighborTables<3>;
```

to the [bottom of `neighbor_tables.hpp`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/render/brep/neighbor_tables.hpp#L277)?

yawor, 2019-08-30:
Thanks, that was it. The same has to be done with EdgeTables too. After that everything builds. Do you want me to prepare PR with this change? Or is there some other solution to fix this if we know what's the reason?

mkeeter, 2019-08-30:
PR please.  I don't think there's a better solution – this is probably Clang vs GCC disagreeing on a corner of the C++ specification.

PR #303 (2019-08-31): Explicit instantation declarations for EdgeTables and NeighborTables templates
Fixes #302
closed 2019-08-31 by mkeeter