libfive/libfive #314: Not obvious export Studio function

Renha, 2019-11-20:
### Description

The "export" menu item is not obvious and counter-intuitive.

I was trying to export .png image of the current scene. I have found nothing similar but the "export" menu item, which didn't state export format anywhere, so of course I thought it is selectable later. But later there was just file select window, without predefined file extension and/or extension list. So obviously I thought that the file format depends on the extension of the file I'd type in, so I used "shape.png". But it doesn't work with image viewer.

I did expect that:
- if only one export format is supported: export function would be clearly labelled (Like "Export PNG image" for example),
- otherwise: program would supply the list of supported export formats to choose from,
- otherwise: program would use the format according to the export file extension.

### Versions

- Operating system: Ubuntu 18.04

mkeeter, 2019-11-20:
The export menu says "Export STL..." in current builds:

[image]

I'm not sure where your build comes from, but if you rebuild from `master`, it should be correct.

closed 2019-11-20 by mkeeter

Renha, 2019-11-20:
Oh! I'm sorry, that was really old revision, `0f517dde9521d751310a22f85ee69b2c84690267`, which I've installed using nix from https://github.com/NixOS/nixpkgs/blob/master/pkgs/development/libraries/libfive/default.nix
I'll update that when I'll have more time. Sorry for troubling!