libfive/libfive #315: reduce visual distractions in studio editor

wluker, 2019-11-22:
This is a neat project. I've been enjoying it, but as my designs get larger I find the red underlining and flashing error messages in the editor distracting. Maybe others don't have this issue, but for me it is a little bit debilitating.

Possible ideas:
1. Turn off the red underline.
2. Allow the user to resize the error window, so it doesn't change size with every key press.

Thanks for this project!

smilberger, 2021-01-14:
I agree it is an issue, normal interaction shouldn't be flagged as an error.

Although possibly not "intuitive" for new users, one idea I had was taking a cue from conventional Lisp/Scheme REPLs, Emacs, etc. and using C-x C-e for "evaluate last expression".  An alternate key sequence could evaluate the entire text box.

And/or, since Studio is a GUI, this functionality could be handled through buttons.

smilberger, 2021-01-17:
Another option is to use the ```Edit > Automatically reload changes``` option in the Studio menu, and edit the file in an external editor.  Each time you save the model in the external editor, Libfive will detect the change and reload the whole file (as the menu item indicates).

Adding the command line argument "--no-syntax" to Studio gets rid of a lot of the red underlining.