libfive/libfive #319: Errors while building on win VS2017

trsh, 2019-12-07:
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2039	'size': is not a member of 'libfive::Intersection<3>'	five	c:\projects\vox\libfive\libfive\src\render\brep\dc\dc_tree.inl	319

trsh, 2019-12-07:
After commenting that line and building solution and running libfive-test.exe

[image]

trsh, 2019-12-07:
And Im not sure where us the actual EXE I should run to open the program.
help!

trsh, 2019-12-07:
Ubuntu 18:04

./libfive-test 

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Cache: performance of commutative tree balancing
-------------------------------------------------------------------------------
/home/trsh/forks/libfive/libfive/test/cache.cpp:411
...............................................................................

/home/trsh/forks/libfive/libfive/test/cache.cpp:439: FAILED:
  REQUIRE( false )
with messages:
  std::chrono::duration_cast<std::chrono::microseconds>(dt) .count() := 964
  (0x3c4)
  std::chrono::duration_cast<std::chrono::microseconds>(dt_) .count() := 14483
  (0x3893)

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
/home/trsh/forks/libfive/libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
sphere                                              1     94559799   94.5598 ms 
Menger sponge                                       1    240941456   240.941 ms 

-------------------------------------------------------------------------------
SimplexTree<3>: assignIndices with cell collapsing
-------------------------------------------------------------------------------
/home/trsh/forks/libfive/libfive/test/simplex.cpp:322
...............................................................................

/home/trsh/forks/libfive/libfive/test/simplex.cpp:349: FAILED:
  REQUIRE( (a.first - b.first).norm() < 1e-9 )
with expansion:
  0.3124999997 < 0.000000001
with messages:
  a.first := -0.625
  0.9375
   0.625
  b.first := -0.625
    1.25
   0.625
  k.first.first->region.lower.transpose() := -0.625  0.625      0
  k.first.first->region.upper.transpose() :=     0  1.25 0.625
  (a.first - b.first).norm() := 0.3124999997

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
/home/trsh/forks/libfive/libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
SOS root                                            1       396384   396.384 µs 

===============================================================================
test cases:    183 |    181 passed | 1 failed | 1 failed as expected
assertions: 886043 | 886041 passed | 1 failed | 1 failed as expected

trsh, 2019-12-07:
I dont understand what it takes to run the software build from source. It Just a test as a result?

mkeeter, 2019-12-07:
This should be fixed in #318 .

I'm not sure what program you're hoping to run (besides the unit tests) – the Studio UI does not currently work on Windows, because it uses Guile Scheme as its embedded language.

trsh, 2019-12-07:
OK Then never mind. I wanted to run studio of course. I managed to do it on Linux.

closed 2019-12-07 by trsh

omgitsraven, 2021-02-06:
I'm having this exact same issue (as the screenshot in the second post), and I'm interested in running libfive on Windows to whatever extent I can (I'm under the impression that I can still make it generate an STL from an .io file on the command-line, right?)

mkeeter, 2021-02-07:
Unfortunately, I don't think that will work – [`export-meshes`](https://github.com/libfive/libfive/blob/master/bin/export-meshes) already exists for headless meshing, but it's still running through Guile Scheme.

This is because the `.io` script has to be interpreted by Scheme to generate the f-rep model, which is then rendered by `libfive`'s meshing algorithms.

omgitsraven, 2021-02-07:
ahh, I misunderstood -- I thought it was just something used for the UI...

In that case, maybe at some point I'll try getting it running on WSL, but in the meantime I'll probably just let it go since I have it working in a VM again for now, heh