libfive/libfive #326: Feature request: remember "show axes/bounding boxes" choice

omgitsraven, 2019-12-27:
Personally, whenever I launch Studio, the first thing I do (after switching to iso-simplex and back to work around that launch bug...) is to turn off the axes and turn on the bounding box... it would be nice if Studio remembered those choices (either per-file or just per-installation, whichever's easier.)

omgitsraven, 2020-01-02:
This should also go for the Z-up/Y-up choice, for that matter.

mkeeter, 2020-01-03:
This definitely seems like a reasonable feature!

I've got a lot of non-`libfive` things on my plate, but if anyone wants to contribute, [`QSettings`](https://doc.qt.io/qt-5/qsettings.html) is the platform-independent abstraction for this kind of persistence.