libfive/libfive #327: Another file that crashes Studio on launch

omgitsraven, 2019-12-27:
It looks like issue #239 isn't totally fixed; the example I provided in that issue no longer crashes on launch, but I've found a very similar situation that performs that same remapping division, which DOES still crash on launch (or at least, as soon as I switch to iso-simplex and back).

```
(set-bounds! #[-10 -10 -10] #[10 10 10])
(set-resolution! 20)
(set-quality! 8)

(define (boxy) (box #[-1.5 0.5 0] #[1.5 3 0]))

(define (other)
  (remap-shape ((boxy) x y z)
    x y (- z  (/ 0.1 y))
  )
)

(blend-rough (sphere 1) (other) 0.3)
```

That's about as simple as I could get it before the crashes stopped happening.

### Versions

- Operating system: Ubuntu 18.04.3 LTS
- `libfive` commit hash: 6e39254e57c179459bb929df49ae96a6017a0ed6
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.14.0

mkeeter, 2019-12-28:
This is probably a division-by-zero issue caused by `(/ 0.1 y)`, which breaks C0 continuity.

It's hard to come up with a *sensible* meshing when things to go ±infinity via NaN, but it still probably shouldn't crash.

omgitsraven, 2019-12-28:
Yeah, I can totally understand if it threw an error or just generated an ugly tangle of polygons, but completely crashing to desktop means potentially losing work, and makes it real hard to debug where the problem lies...