libfive/libfive #329: Y-up mode is actually X-up!!

omgitsraven, 2020-01-02:
I finally got around to trying y-up mode, and, umm.... well, the title says it all :)

Here's a little test scene if you need a demonstration...

```
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 10)

(define (letterX)
  (union
    (sequence
      (box-centered #[1 4 1])
      (rotate-z (/ pi 4))
    )
    (sequence
      (box-centered #[1 4 1])
      (rotate-z (/ pi -4))
    )
  )
)

(define (letterY)
  (union
    (sequence
      (box-centered #[1 2 1] #[0 1 0])
      (rotate-z (/ pi 4))
      (symmetric-y)
    )
    (box-centered #[2 1 1] #[1 0 0])
  )
)

(define (letterZ)
  (union
    (box-centered #[3 1 1] #[0 1.5 0])
    (box-centered #[3 1 1] #[0 -1.5 0])
    (sequence
      (box-centered #[3.5 1 1])
      (rotate-z (/ pi 4))
    )
  )
)

(move (letterX) #[5 0 0])
(move (letterY) #[0 5 0])
(move (letterZ) #[0 0 5])
```

never mind the funny orientations of the letters; their positions are the point.

In Z-up mode, the axis that can spin forever pivots around the Z...
but in Y-up mode, the axis that can spin forever pivots around the X!

omgitsraven, 2020-01-02:
Figured this one was a simple enough substitution that I could fix it myself even if I don't much understand the C++ side of things, so I did :) b6f15ad4e6ff49ce5041671ad7389eaa12b8b156

I'm a little confused why the old quaternion wasn't right (given that it changed the look axis from Z to Y), or why changing the up axis from X to Z fixed it (I only found it after trying a few combinations after my initial assumptions didn't work out), but... for now it seems to work? (And if nobody's reported it yet, I guess there's nobody who would be depending on the old behaviour?)

Anyway, I do feel a *little* nervous committing a fix that I don't entirely understand, so feel free to tell me if I should revert it until you've identified exactly what was going on, but I'll close my issue for now.

closed 2020-01-02 by omgitsraven

mkeeter, 2020-01-03:
The traditional fix for any graphics issue is to poke at matrices until things *look right*, so I think this is fine 😛 – thanks for the patch!