libfive/libfive #330: How to start using? [Tutorials]

iasj, 2020-01-15:
I couldn't find any tutorial anywhere. I use OpenSCAD and they have an extensive documentation. Have you?

mkeeter, 2020-01-17:
Nope!  What are you trying to do?  Have you succeeded in opening Studio and seeing the example file in there?

iasj, 2020-01-17:
I'm not in need of anything in particular as in right now. It is just that I've using OpenSCAD for quite a while and I use it for work (3D design and printing).

iasj, 2020-01-17:
I'm not in need of anything in particular as in right now. It is just that I've been using OpenSCAD for quite a while and I use it for work (3D design and printing).

Although OpenSCAD has been doing fine for now, I always enjoy having another source of free software of the same kind. 

I also think of writing my own CAD software and I don't wanna start from scratch. I checked that libfive has API for C. That is cool, and wanna know more about it.

About your question now. I installed it in my Arch Linux via AUR package. It went fine, it opens the Studio and its example file, but I just don't know what to do next. I mean, I guess libfive lacks usage documentation.

I wanna learn the syntax and its details just like I did with OpenSCAD from its website.

s-ol, 2020-02-09:
I've read the tutorial file and the other examples, but I can't figure out how to destructure a vector (or whatever the `[1 2 3]` syntax creates, it doesnt seem to exist in regular Guile) to make my own shapes that are ergonomic to use.

mkeeter, 2020-02-09:
Yup, that syntax creates a `vec2` or `vec3`.  You can get components with the `.x/y/z` functions, e.g.
```scheme
(define v [1 2 3])
(.x v) ; returns 1
```

nyaray, 2020-04-18:
I just spent a good 30-45 minutes in Studio, figuring out that objects that go beyond the bounds result in nothing being drawn. I don't know exactly where, but a line about that would be nice. Maybe a caveats-section somewhere?

Please let me know if this would count as issue-jacking or not, I don't mind creating a separate issue for "getting started hints/suggestions" for Studio, specifically.