libfive/libfive #334: Trying to build for Ubuntu 19.10, bit confused about Qt path

mderouss, 2020-02-11:
### Description

[Description of the issue]

I'm using Ubuntu ( Xubuntu in fact ) 19.10. I've installed the dependencies given in the instructions. But I'm confused about what is meant by \<Qt Install Path\>. The qtbase5-dev install does not appear to create any path that looks like the one given. Here is some information about the qt installation -

> mark@z800:/usr/lib/x86_64-linux-gnu/qtchooser$ qtchooser -print-env
> QT_SELECT="default"
> QTTOOLDIR="/usr/lib/qt5/bin"
> QTLIBDIR="/usr/lib/x86_64-linux-gnu"
> mark@z800:/usr/lib/x86_64-linux-gnu/qtchooser$ qmake -v
> QMake version 3.1
> Using Qt version 5.12.4 in /usr/lib/x86_64-linux-gnu

There are various other parts of QT scattered around the system. I won't list every directory, but none of them have the form given in the instructions. Can someone explain what  \<Qt Install Path\>/5.7/gcc_64 is actually supposed to refer to ?  Presumably the 5.7 is just the QT version. gcc_64 looks like a reference to a compiler. gcc is certainly installed -

> mark@z800:~/Projects/libfive$ gcc -v
> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
> OFFLOAD_TARGET_NAMES=nvptx-none:hsa
> OFFLOAD_TARGET_DEFAULT=1
> Target: x86_64-linux-gnu
> Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.2.1-9ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
> Thread model: posix
> gcc version 9.2.1 20191008 (Ubuntu 9.2.1-9ubuntu2) 

[Steps to reproduce]

Install the dependencies as listed and try to follow the instructions.

[What did you expect to happen?  What happened?  All the time?]

The instructions are unclear. No build was performed.

[Are there any error messages printed to the console?]

No

### Versions

- Operating system: Ubuntu ( Xubuntu ) 19.10
- `libfive` commit hash: aa32adbc434242b12d855b707b8b1f014d0b170d
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.12.4

omgitsraven, 2020-02-11:
I had the same problem; I'm not a Linux or Qt expert, but after a lot of googling and some experimentation, the solution that worked for me was `-DCMAKE_PREFIX_PATH=/opt/Qt/5.14.0/gcc_64`

mderouss, 2020-02-11:
I found a similar reference. But there is no such directory in my system - it's not a directory that the qt5base-dev install creates. Did you create this directory, or was it already there ?

omgitsraven, 2020-02-11:
I honestly don't remember, unfortunately; I might have tried manually installing Qt a few times through various package managers.

Just to be sure—it's not just a version difference, right? i.e. is there an /opt/Qt/ folder with a different version number folder inside that?

mderouss, 2020-02-11:
There's no /opt/Qt folder at all. My guess is that in fact, in my environment,  there is a missing dependency, but Qt seems to be horrendously complicated, and I've no idea what that dependency might be. I'd like to understand something very basic though - is this directory an output directory that the build writes to, or does the build expect to find something specific already there ? That would be good to know. Thanks for the tip though, nice to know someone is out there :).

omgitsraven, 2020-02-11:
Ah! Good question—Qt is a library (a UI library specifically) that the compiler is expecting to *find* there, because Studio needs it in order to put a window on the screen, basically.

mderouss, 2020-02-11:
OK, understand now.

CMAKE_PREFIX_PATH is indeed a path that points various cmake functions at preexisting things. cmake will use this path to look for programs, libraries, etc. ( see https://cmake.org/cmake/help/latest/command/find_library.html#command:find_library for example ). So CMAKE_PREFIX_PATH is expected to be the root of the Qt installation, it's not an output directory. 

On my system, it appears that the root of the Qt installation is /usr, rather than a version specific directory in /opt. These options are described here : http://www.linuxfromscratch.org/blfs/view/svn/x/qt5.html , where /opt is considered to be the preferable technique. 

However, when using /usr I need to ensure that CMAKE_LIBRARY_ARCHITECTURE is set to x86_64-linux-gnu, because otherwise some Qt things ( include files for example ) may not be found.

mderouss, 2020-02-11:
I've now built libfive/studio successfully, so I'm closing this. I didn't need to set CMAKE_LIBRARY_ARCHITECTURE, though I'm not clear why.  

Just one more point about the instructions though. The 'make' command given specifies '-j4'. If you are using gnu make ( you probably are, on Linux ), this will have no effect, multiple jobs will not happen, and you really need to use '-j 4', i.e. there's a space between the j and the 4.

closed 2020-02-11 by mderouss