libfive/libfive #336: Guile Module install (and install separately to base libfive)

KoviRobi, 2020-04-07:
Hi,
### Description
I am installing libfive on Nix, and it doesn't seem to install the guile modules. I think it just does the basic cmake install steps (cmake flags: "-DCMAKE_BUILD_TYPE=Release"), any flags I need to give it to install the .scm/.go files?

Also, can I make a package for just libfive, then compile libfive-studio/libfive-guile using the existing libfive package (it will make the separate licenses easier to express).

### Versions

- Operating system: NixOS (https://github.com/KoviRobi/nixpkgs/tree/libfive-2020-02-15)
- `libfive` commit hash: 5b7717a25064478cd6bdb190683566eaf4c7afdd
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.12

KoviRobi, 2020-04-12:
Ah, sorry, I have found https://github.com/libfive/libfive/blob/master/doc/guide.md which does help, turns out I should have done
```
(load-extension "libfive-guile" "scm_init_libfive_modules")
```

KoviRobi, 2020-04-12:
Actually, perhaps there should be a `libfive/modules.scm` or something, e.g. https://www.gnu.org/software/guile/manual/html_node/Modules-and-Extensions.html#index-extensiondir-1 ?

mkeeter, 2020-04-13:
Perhaps!  The Guile module is pretty specialized for the Studio GUI right now, so less work has been put into making it friendly as a general-purpose module.  If you want to open a PR that brings up to best practices, I'd be happy to merge.

mkeeter, 2021-02-20:
Okay, this is done in #383 (along with a bunch of other changes).

closed 2021-02-20 by mkeeter