libfive/libfive #338: Importing meshes or vectors?

traverseda, 2020-05-06:
Is there any way to import STL or DXF files into libfive? I imagine that it would be pretty challenging to convert them into whatever internal f-rep data structure libfive uses.

mkeeter, 2020-05-07:
Not yet!

It's definitely a tricky problem.  The approach would be to embed them as a black-box oracle (which `libfive` already supports) but writing that oracle would be hard.

`libigl` has a [`point_mesh_squared_distance` function](https://github.com/libigl/libigl/blob/master/include/igl/copyleft/cgal/point_mesh_squared_distance.h), which does part of the work, but
- doesn't work for intervals 
- depends on CGAL, so dunno about license implications.

SamuelAl, 2021-11-24:
Hi @mkeeter ,
What operations would the black-box oracle have to support? Do you have any idea how someone would begin implementing this? Interested in the subject and would love to see if you had any thoughts on this.

mkeeter, 2021-11-24:
I talked about the Oracle API in detail [here](https://github.com/libfive/libfive/discussions/468) (from the context of integrating a _different_ SDF toolkit, but the API would be the same for meshes).

Doing a proof of concept with `libigl` is reasonable!

It will be pretty slow without interval support, i.e. "what's the conservative `[min,max]` distance to the mesh within this 3D spatial region?", but it should still work.

This also won't preserve sharp features well without a way to compute the gradient, but that could come later.

SamuelAl, 2021-11-25:
Thank you! Seems like a very good starting point.

Ammarkhat, 2022-02-19:
@SamuelAl did you try to implement this?