libfive/libfive #340: Possible cone replacement

omgitsraven, 2020-05-07:
I needed to do some work that would involve expanding a cone (via `clearance` specifically) but upon trying, I noticed that it doesn't really offset well at all:

[image]

I went to add a new `cone-exact` function based on the one here: https://iquilezles.org/www/articles/distfunctions/distfunctions.htm but in the process, I noticed that while it offsets much better...

[image]

...it isn't actually "exact" at all, since the distance offsets correctly on the sides, but not on the corners:

[image]

An exact distance function's expansion would get rounder at those corners, right? (The bottom corners are my fault because I'm just clipping it with a half-space, but the top corner isn't right either…)

But anyway, even if it's not exact, it already seems a lot more suitable for at least this purpose, than the existing one... Do you think it would be a good idea to replace the existing `cone` function with this one?

```
(define (cone-angle-exact c)
  (lambda-shape (x y z)
    (dot
      c
      #[ (norm #[x y]) z ]
    )
  )
)
(define (cone-exact r h m)
  (let ((ang (atan r h)))
    (sequence
      (cone-angle-exact #[(cos ang) (sin ang)])
      (move #[0 0 h])
      (difference (half-space #[0 0 1]))
      (move m)
    )
  )
)
```
(I've called it `cone-exact` here just because that's what I was trying to make, heh...)

I guess it's a bit more complex than the existing one, but I'm not sure if it's prohibitively so? (Or if maybe it has other downsides that I just haven't encountered yet…)

Anyway, as it happens, his capped-cone function DOES seem to be exact (at first glance anyway), and can be used as a cone function by setting the second radius to 0, so maybe I'll add that as well, and then add `cone-exact` as a wrapper for it with one less argument (the second radius being hard-coded to zero)?

mkeeter, 2020-05-07:
Yeah, that looks much nicer than the existing cone, which has a nasty singularity at the tip IIRC.  

I'd support making that the basic `cone`, then having `cone-exact` be a capped-cone with second radius 0, like you suggested.

(In general, replacing my distance functions with IQ's is probably always a good idea 😆 )

omgitsraven, 2020-05-07:
Sounds good :) I've hit a snag implementing the capped-cone one without conditionals (which I'm writing its own thread about now!) but I'll get the other cone in in the meantime :)

and yeah, oh my gosh, I forgot about that plane that has a habit of appearing at the tip of the cone (even though I just ran into it yesterday!)—hooray for two birds with one stone…

Will get this in soon, I'm kind of a few things deep in a problem-solving stack right now, hahah

omgitsraven, 2020-05-07:
OK so: I asked IQ about the cone that was described as exact when it's not, and he updated his page with:
1. a properly exact one, which is great but I can't implement it right now because it depends on ternaries (or at least `sgn`)
2. an updated version of the non-exact one I demonstrated above (now labeled as non-exact) which has a height instead of being infinite, so I can skip doing the difference on a half-space!

Anyway, I'm not sure if you have any particular insight about what's better between the one above and this new one:

```
(define (cone-angle-new-alt c h)
  (lambda-shape (x y z)
    (let ((q (norm #[x y])))
      (max
        (dot c #[q z])
        (- (- h) z)
      )
    )
  )
)
(define (cone-new-alt r h m)
  (let ((ang (atan r h)))
    (sequence
      (cone-angle-new-alt #[(cos ang) (sin ang)] h)
      (move #[0 0 h])
      (move m)
    )
  )
)
```
but I think at least for now I'll go for the new one, because IQ said so ;)

omgitsraven, 2020-05-07:
hang on; I realised I can just cancel out the h shift by putting it somewhere else:
```
(define (cone-angle-new-alt a h)
  (lambda-shape (x y z)
    (let ((q (norm #[x y])))
      (max
        (dot #[(cos a) (sin a)] #[q (- z h)])
        (- z)
      )
    )
  )
)
(define (cone-new-alt r h m)
  (sequence
    (cone-angle-new-alt (atan r h) h)
    (move m)
  )
)
```
so unless I've made a mistake somewhere (it seems fine??) then this is probably the one to go with :)

omgitsraven, 2020-05-08:
Done! e1da2fb6c2171a399c9a109e99e83d704310a6e2

closed 2020-05-08 by omgitsraven