libfive/libfive #341: Shape-safe ternaries?

omgitsraven, 2020-05-07:
So, I know that we can't use conditionals inside `lambda-shape`. But there are workarounds for some simple operations; in a previous thread, you suggested using `(lambda-shape (x y z) (max x 0))` to replace `(lambda-shape (x y z) (if (> x 0) x 0) )`, for example.

What I'm hoping for now is something more general that will allow for ternaries that return one of two values based on a *third* value being above or below zero, i.e. `(if (< x 0) a b)`... in theory, it should be totally possible, as long as there's a function that returns either 0 or 1 based on something else being above zero, i.e.

```
(define (faketernary-ltz x a b)
  (+
    (* a (zeroOrOne-ltz x) )
    (* b (zeroOrOne-ltz (- x) )
  )
)
```
and in theory, the zeroOrOne-ltz function could be something like
```
(define (zeroOrOne-ltz x)
  (max 0 (/ (abs x) x))
)
```
…if it weren't for the fact that calling it with EXACTLY zero causes a numerical overflow from dividing zero by zero!

So I guess my question is any one of:

1. is there a way to catch a numerical overflow and safely coerce it to zero
2. is there any shape-safe function that would do something like return 1 for an input of 0 and 0 otherwise, that we could wrap the divisor in so that zero has something safe to divide by
3. can you think of a better way to normalize all inputs to either 1 or 0, without using division?
4. is there some much simpler shape-safe way of doing this altogether (or could you add one at some lower level?)

(as a footnote, I just noticed that my `faketernary-ltz` also has a problem at zero, but if we could solve or replace the zero problem, we could wrap that ternary in another ternary that checks for exactly zero with a zero-or-one fake conditional the same as `zeroOrOne-ltz` without the `abs` around the numerator... but this is starting to get awfully verbose for something that it feels like there MUST be SOME other way of expressing!)

The closest I've come so far is this:
```
(define (zeroOrOne-ltz x)
  (max 0 (ceiling (/ x 999999999)))
)
```
(assuming 999999999 gets replaced with some safe max_int value, if such a thing exists in scheme) which gives me what I want for trivial inputs, but feels uncomfortably prone to precision problems...

I don't have a great understanding of the nature of the JIT-ish language that lambda-shapes get transformed into in libfive, but this FEELS like something that should fit within its constraints! "Return 1 if 0 and vice versa" is an operation that just takes one number and returns another, and plenty of supported functions do that; would it be possible for you to hard-code one that does that, to make these shape-safe ternaries possible?

mkeeter, 2020-05-08:
There's actually a more fundamental reason against having ternary operators – the algorithms for rendering and meshing are pretty flexible, but require C^0 continuity, i.e. `f(x + ε, y, z) == f(x - ε, y, z)` should be true as `ε` goes to 0.

Otherwise, we end up with awkward situations where a point is both inside and outside the shape (the simplest case is `1/x` as `x` crosses 0, which is handwavingly equivalent to `±inf`).  

(As a side note, this is the cause of most meshing segfaults, e.g #327, because `(- z  (/ 0.1 y))` can break continuity.  Obviously, things shouldn't segfault, but being robust against these cases is hard!)

One building block to consider is a sigmoid-ish function, e.g. `x / sqrt(1 + x^2)`.

This is well-defined everywhere, not too bad in terms of intervals, and smoothly blends between -1 and 1.  The downside is that it has a user-defined radius, so it's not independent of scale (though you could obviously change the transition length by multiplying `x`).

You could use this to build your pseudo-ternary operator, which will *blend* the two shapes based on the condition (rather than selecting one vs the other with discontinuities).

omgitsraven, 2020-05-08:
I mean, I understand how they can be misused in general, but the exact cone function that I need to use them in certainly seems to be producing smooth results, at least in my raymarching tests—I think it's being used to stitch together two functions whose position and slope meet at a certain point, so it should be safe. (And like you said, there's already "safe" functions that aren't, like division, or even `ceiling`...) I don't think blending would work though, because who knows what both those functions are doing on the opposite sides of the clipping point that would suddenly be revealed if they were blended rather than spliced?

Hopefully I can rig something up out of sigmoid functions and round/ceil/floor; it still seems like an enormous amount of extra math compared to a simple 1-or-0 operator though.

doug-moen, 2020-05-09:
RavenWorks, you are probably using the same exact cone distance function that I'm using in Curv. Here it is:
```
dist[x,y,z,t] =
    let q = [mag[x,y], z]; 
        apex = q - [0, height];
        mantleDir = normalize[height, radius];
        mantle = dot[apex, mantleDir];
        projected = dot[apex, [mantleDir[Y], -mantleDir[X]]];
        d = max[mantle, -q[Y]];
    in do
        // distance to apex
        if (q[Y] > height && projected < 0)
            d := max[d, mag(apex)];
        // distance to base ring
        if (q[X] > radius && projected > mag[height, radius])
            d := max[d, mag(q - [radius, 0])];
    in d;
```
The result, `d`, is a continuous function of `[x,y,z]`. It's fine for raymarching. But LibFive works differently: all of the intermediate results must be continuous functions, not just the final result. Conditional expressions are "unsafe" because they often return a discontinuous function of their inputs. Matt is banning an unsafe operation that could be easily misused, causing the evaluator to foul up.

In this particular case, the `if` statement is guarding a statement of the form
```
d := max[d, ...];
```
And I think this is safe for libfive. I don't have a mathematical proof of this on hand, but we know that the variable `d` was a continuous function before being conditionally modified by the `if` statement, and we know that `d` is continuous after being modified.

Matt: In principle, I think that LibFive could provide an if_max(cond, a, b) function that is safe for the evaluator. Does this make sense to you?

mkeeter, 2020-05-10:
@doug-moen That's an interesting function – I have to think about it a little more.

> But LibFive works differently: all of the intermediate results must be continuous functions, not just the final result.

This part isn't right – intermediate results can be discontinuous, it's just the final shape that must be C^0 continuous for meshing.

This discussion has reminded that I actually *did* implement a conditional operator, way back in the day:

```
(compare a b) => -1 if a < b
                  1 if a > b
                  0 otherwise
```

I didn't end up using in many places, because of the segfaulting-if-you-screw-up behavior.

(...one hour later...)

In fact, I've gotten tremendously nerd-sniped, and ended up implementing IQ's latest `Cone - exact` code:

[image]

This is using `compare` as a replacement for `sgn`.  It's not 100% identical, because of the behavior at 0, though it's not hard to build `sgn` from it.

Here's the code:
```(set-bounds! [-10 -10 -10] [10 10 20])
(set-quality! 8)
(set-resolution! 10)

(define (exact-cone a h)
  (lambda-shape (x y z)
    (let*  ((c [(sin a) (cos a)])
            (q (* h [(/ (.x c) (.y c)) -1]))
            (w [(norm [x z]) y])
            (a (- w (* q (clamp (/ (dot w q) (dot q q)) 0 1))))
            (b (- w [(* (.x q) (clamp (/ (.x w) (.x q)) 0 1))
                     (.y q)]))
            (k (compare (.y q) 0))
            (d (min (dot a a) (dot b b)))
            (s (max (* k (- (* (.x w) (.y q))
                            (* (.y w) (.x q))))
                    (* k (- (.y w) (.y q))))))
            (* (sqrt d) (compare s 0.0))
)))
(apply union
  (map (lambda (i) (move (- (exact-cone 0.4 2) (/ i 5))
                         [0 0 (* i 4)]))
       (iota 4)))
```
@omgitsraven want to poke this a bit more and see if it works for you?

(One more subtle downside of `compare` is that using it in this way doesn't prune expressions out of the tree as efficiently, but that's a discussion for another day)

omgitsraven, 2020-05-10:
Ahh, wonderful! I'm not sure when I'll next have a chance to look at this, but this seems very helpful, thank you :D

omgitsraven, 2020-05-23:
Was just getting ready to put this into the library, when I ran into a fantastic bug:

[image]

Run this to see it:

```
(set-bounds! [-1 -3 -1] [3 3 4])
(set-quality! 8)
(set-resolution! 20)

(define (cone-exact-ang a h)
  (lambda-shape (x y z)
    (let*  ((c [(sin a) (cos a)])
            (q (* h [(/ (.x c) (.y c)) -1]))
            (w [(norm [x y]) z])
            (a (- w (* q (clamp (/ (dot w q) (dot q q)) 0 1))))
            (b (- w [(* (.x q) (clamp (/ (.x w) (.x q)) 0 1))
                     (.y q)]))
            (k (compare (.y q) 0))
            (d (min (dot a a) (dot b b)))
            (s (max (* k (- (* (.x w) (.y q))
                            (* (.y w) (.x q))))
                    (* k (- (.y w) (.y q))))))
            (* (sqrt d) (compare s 0.0))
)))
(define (cone-exact r h)
  (cone-exact-ang (atan r h) h)
)

(move (cone-exact 0.4 2) #[0 0 2])
```

(It's another one of those ones that comes and goes with different bounds.)

mkeeter, 2020-05-23:
Wow, that's quite a glitch!  I wonder if there are numeric stability issues somewhere in the cone function...

lf94, 2023-06-14:
Just documenting this for the next souls that come across this:

I ran into this exact problem last night, and while I was able to find an equivalent to `a < n ? b : c` on my own in the past using just compare, min and max - I was not able to do it for Inigo's rounded capped cone. It became too much for my peanut brain.

Now that I've read this issue page, I understand now that it's actually best to use the SDFs which do not have these compares, even if they aren't exact. I've resorted to using `taper_xy_z`, `cylinder` and `offset` to create my rounded capped cones, which I'm sure libfive can optimize many operations away.