libfive/libfive #342: Get sign of number?

omgitsraven, 2020-05-07:
This might be superceded by #341, but in the meantime: is there a (shape-safe!) function to return the sign of a number? This page https://stackoverflow.com/questions/8782685/sign-function-in-scheme suggests that it exists and is called `sgn`, but libfive doesn't seem to recognize it…

mkeeter, 2020-05-08:
I gave more context in #341, but I'd suggest playing with sigmoid functions!

The [Wikipedia page](https://en.wikipedia.org/wiki/Sigmoid_function) lists a bunch.  From quickly eyeballing the list, `x / sqrt(1 + x^2)` is probably the cheapest to evaluate.

mkeeter, 2020-05-10:
As I rediscovered in #341, this can be built pretty easily with the existing shape-safe `compare` operator:

```
(define (sgn a)
  (compare a 0))
```

closed 2020-05-10 by mkeeter