libfive/libfive #343: Revolve `x0` not used?

KoviRobi, 2020-05-15:
Perhaps I'm being silly, but the `x0` optional argument isn't used in revolve

https://github.com/libfive/libfive/blob/e1da2fb6c2171a399c9a109e99e83d704310a6e2/libfive/bind/transforms.scm#L380-L387

KoviRobi, 2020-05-15:
I also wonder how do can I make a corkscrew shape, e.g.
```
(define* (revolve-y shape #:optional (x0 0))
  "revolve-y shape [x0]
  Revolves a 2D (xy) shape around the vertical line x=x0"
  (union
    (remap-shape (shape x y z)
      (sqrt (+ (square x) (square z))) (- y (atan x z)) z)
    (remap-shape (shape x y z)
      (- (sqrt (+ (square x) (square z)))) (- y (atan x z)) z)))
(revlove-y (circle 1 #[10 0]))
```
doesn't produce anything neat (probably because of the discontinuity, but then how do I make it infinite?)

mkeeter, 2021-04-26:
This got ported to the C++ standard library, and now [uses `x0`](https://github.com/libfive/libfive/blob/master/libfive/stdlib/stdlib_impl.cpp#L538-L544)!

closed 2021-04-26 by mkeeter