libfive/libfive #344: Does `pow` work in studio?

kayarre, 2020-05-21:
### Description

When using studio it doesn't appear to allow the pow command. e.g.

```
(define R0 1.4 )
(define L (* R0 2.5 ))

(define-shape (ellipsoid x y z)
  (let* ((r (+ (pow (* x (/ R0 L)) 3) (square y) (square z)))
         (theta (atan y x))
         (phi (acos (/ z r)))   
         )
    (- r (square R0))))

ellipsoid
```

I guess I was expecting to have a `pow` command

```stack trace: unbound-variable: Unbound variable: pow```

### Versions

- Operating system: Linux mint 19.3
- `libfive` commit hash: e1da2fb6c2171a399c9a109e99e83d704310a6e2
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): N/A

omgitsraven, 2020-05-21:
In scheme, it's `expt`.

closed 2020-05-21 by omgitsraven

kayarre, 2020-05-22:
Thank you I am new to scheme.