libfive/libfive #349: Help with C++ api example project.

kurtsansom, 2020-06-09:
### Description
I have made a cool thing with the studio app, but now I want to convert that to using the lower level C++ interface.

The example in the documentation kinda covers it, but do you an example using the c++ interface that creates and object and exports the stl based on some parametric input  and has it's own cmake file that builds and links with library?

I know this is simple, but wanted to see if you had an example already around. 

The Part I am getting stuck on is with cmake I typically use the find_package command, and am not sure how to deal with the pkgconfig files.

Thank you so much for sharing your work, and your time and effort.

mkeeter, 2020-06-09:
Congrats on the cool thing – is it going to be made public eventually?

The easiest way to build on `libfive` right now is to include it as a Git submodule, then use `add_subdirectory(libfive)` to include it in the build.  You can also use `set(BUILD_STUDIO_APP OFF)` in your top-level `CMakeLists.txt` (before `add_subdirectory`) so that the submodule only builds the core libraries and not the UI.

You can see an example of this in the [`mpr` repo](https://github.com/mkeeter/mpr/) [here](https://github.com/mkeeter/mpr/blob/master/CMakeLists.txt#L5-L11).

kurtsansom, 2020-06-10:
It’s pretty simple, and yes I would be happy to share it.

Thank you for the input and direction.
-Kurt

kayarre, 2020-06-12:
It looks like the C++ api example is out of date from the current API?

I am having a hard time figuring out a simple example of a sphere and the steps required to generate the shape and export to stl.

FunMiles, 2020-06-12:
I got a simple example running with the C++ API example. It does run but I have not visualized the result. I will do visualization in my own project.

```cpp
#include <iostream>
#include <Eigen/Eigen>
#include <libfive.h>
#include <libfive/render/brep/mesh.hpp>
#include <libfive/render/brep/settings.hpp>

using namespace libfive;
namespace Kernel = libfive;
int main()
{
	auto x = Kernel::Tree::X();
	auto y = Kernel::Tree::Y();
	auto z = Kernel::Tree::Z();

	// Arithmetic is overloaded for the Kernel::Tree type
	auto out = (x * x) + (y * y) + (z * z) - 1;

	// Pick the target region to render
	auto bounds = Region<3>({-2, -2, -2}, {2, 2, 2});

	// Mesh::render returns a unique_ptr, so it cleans up automatically
	Mesh::render(out, bounds, BRepSettings{})->saveSTL("sphere.stl");

	return 0;
}
```

Note: I built the library separately and link to it. There is a big difficulty with the aligned allocator of Eigen. This allocator can default to the standard allocator depending of the compilation flags.
I recommend you add `-D__AVX__ -march=native` to your compilation flags to be compatible with the ones the library will trigger from its CMakeLists.txt.

FunMiles, 2020-06-12:
PS: I, like @kurtsansom am also for having an install that creates a proper CMake find_package compatible file. It's not terribly difficult to do with the recent CMake versions. I may create one if I have the time and will the do a PR.

kayarre, 2020-06-12:
I couldn't get  my example to work, I think I was missing the namespaces but I did get this to work.

```
/*
C++ example
*/
#include "libfive.h"

int main(int, char**)
{
    auto x = libfive::Tree::X();
    auto y = libfive::Tree::Y();
    auto z = libfive::Tree::Z();

    auto out = (x * x) + (y * y) + (z * z) - 1;
    libfive_region3 bounds = {{-2, 2}, {-2, 2}, {-2, 2}};

    std::cout << libfive_tree_print(&out) << std::endl;
    bool result;
    result = libfive_tree_save_mesh(&out,
                        bounds,
                        10,
                        "test.stl");
    std::cout << " did it work: " << result << std::endl;
}
```

kayarre, 2020-06-12:
I decided to hack it a bit, and I think I have something that works. I will create a merge request in a bit mostly for as a way to get it out, but it has no polish. I am not a cmake expert. I have to relearn it every time I do it.