libfive/libfive #356: Feature request: Blender bindings?

omgitsraven, 2020-07-05:
I know this is a colossal can of worms for a two-word feature request, but: I have to bring it up :) I've been daydreaming about it for a while, but I don't know a thing about making Blender plugins; I guess I'm mostly just curious whether anyone else has been considering it, or if there's any particular reasons why it would or wouldn't be a good fit...

Besides just the convenience of being able to have an editable SDF alongside meshes made by other methods, and being able to apply shaders to the mesh while you work on it, I'm also thinking about benefits like having something like a `(nodehandle shape)` function, which would plop a node in the GUI that could be translated/rotated/scaled in the usual manner, and which would pass its resulting transform matrix back into the scheme…

mkeeter, 2020-07-05:
It's not out of the question: there's already [Unity binding](https://github.com/zalo/libfive-unity), which could offer some inspiration (though I don't know much about the plugins system for either platform).

I won't be putting this on my own plate, but am happy to support if anyone wants to take it up!

joseph-peano, 2020-07-28:
FWIW, some ideas on this topic:

1. It looks like only Python scripts can be used for extending Blender outside of its source tree.

From the Blender [scripting documentation](https://docs.blender.org/manual/en/latest/advanced/scripting/introduction.html):
> Python scripts are a versatile way to extend Blender functionality. Most areas of Blender can be scripted, including animation, rendering, import and export, object creation and automating repetitive tasks.
>
> To interact with Blender, scripts can make use of the tightly integrated API.

From Blender [AntiFeatures](https://wiki.blender.org/wiki/Reference/AntiFeatures)
> Support for Other Scripting Languages
>
> Supporting multiple languages is not something we're currently interested in.
>
> Embedding a language runtime has a significant maintenance overhead, so there would need to be very compelling reasons (besides developer preference) to add support for other languages or to move away from Python.
>...
> C/C++ Plugin API
>
>At the moment we are not planning to accept binary plugins in Blender (which implies a C/C++ API).

2. There is a start to a [Python binding for Libfive](https://gitlab.com/rcmz0/libfivepy).  This could be combined with the Blender Python scripting system to do modeling in Python.

3. [Far-fetched, but conceptually possible]  Building on (2), Peter Norvig wrote a [Lisp interpreter in Python](https://norvig.com/lispy.html).

Combine that with the Python Libfive binding to create models with similar syntax to Guile in the Blender Python system.

There is also a project called Make A Lisp (MAL), which has a [Lisp interpreter written in Python](https://github.com/kanaka/mal/tree/master/impls/python).  Make A Lisp is a nice tutorial-based project to build Lisp implementations in many different languages.  Obviously speed would be a factor, but Libfive is doing the heavy lifting in C++.

4. Use a Blender script to shell out to Libfive, render to an STL file, and import the result.

5. Ignore (1) and talk to the Blender developers about what it would take to integrate Libfive into the code base.  See if they would add it if someone developed it.  They are not likely to add Guile as a dependency, but an embedded mimimal Lisp (e.g. MAL) written in C/C++ could be used.

smilberger, 2020-12-16:
I had a look at the Unity binding for inspiration, and have been able to generate some simple geometry in the Blender Python console window using the existing Libfive Python binding code.  But so far what I have done is only a minor convenience over just importing an STL exported from Studio.  I think the real power would be feedback from the GUI, like in Unity, and I'm looking at the feasibility of that from Blender Python.  Blender did support plug-ins in C++ up until about 2.4, but now it looks like Python only or convincing the maintainers to accept a PR.