libfive/libfive #358: [Help needed] Import image.

warycat, 2020-08-13:
I have some very good feedback using libfive vs openscad. It is very fast and the scheme language makes a lot of sense. One thing I notice is that there is no support of importing an image. Correct me if I am wrong.

mkeeter, 2020-08-14:
You're not wrong!

Out of curiosity, do you want to import an image and see it in the GUI (e.g. if you're trying to match its dimensions), or use it as a basis for solid modeling?  In the latter case, how do you imagine converting the image to a solid?

warycat, 2020-08-14:
The idea comes from a joke about making coasters that people can print from their 3d printers. The image should be imported and converted to a boolean array or a function that maps (x: usize, y: usize) -> bool. Then people can apply linear extrude function on it. In the future, people can open their mobile app that build on top of libfive and import photo from their camera roll. They can do a little coding or drag and drop some shap to create their own artwork.

robryk, 2020-11-01:
I second this request, especially the special case of importing _vector_ graphics.

In my case, I'd like to be able to import an svg file (and get a function that ignores z, and if positive within the interior of closed paths from that file, or something like that). My usecase is importing a 2D shape created using a constraint-solving CAD tool (e.g. Solvespace) or (in some indeterminate future) a shape derived from a generated PCB layout. I might (no guarantees though) implement a svg path specification -> libfive-compatible predicate function in Scheme in the near future.