libfive/libfive #359: crazy edge effects

John-Colvin, 2020-09-08:
I am attempting to make helical shapes, so I am rotating around the y axis while also using the angle of rotation to determine an offset.

Code is as follows

```
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 10)
(set-resolution! 100)

(define (revolve-step shape)
    (remap-shape (shape x y z)
      (sqrt (+ (square x) (square z)))
      (+ y (/ (atan x z) (* 0.5 pi)))
      z
      )
)

(revolve-step (move (circle 1) [2 0 0]))
```

While the single rotation of the helix works nicely, the ends have crazy near-vertical "wings" sticking out.

I'm guessing this sort of approach just isn't possible with the libfive, is there another way I can try?

### Versions

- Operating system: linux
- `libfive` commit hash: 8ca1b86
  - Is this the latest commit to `master`?  [yes]

mkeeter, 2020-09-09:
I believe this is because of the branch cut in `atan`, where it breaks continuity and goes from π to -π (with Z negative, X going from positive-to-negative):

[image]

`libfive` requires the input functions to be C0-continuous, because otherwise the concept of inside-vs-outside is meaningless – `(atan 0 -1)` is ambiguously positive or negative, so meshing becomes unhappy.

I'll ponder whether there's a good way to avoid this, but hopefully knowing the root cause will help you think about it as well!

John-Colvin, 2020-09-09:
I guess the simplest solution is to define an approxAtan that is continuous but very steep in the `x < 0`, `y ≈ 0` region, which would be good enough for most real uses

stylon, 2020-11-27:
I just learned about libfive coming from OpenSCAD and like the concept. @John-Colvin: did you try to visualize with a different meshing algorithm? With "Iso-simplex" it looks perfect:

[image]