libfive/libfive #362: Summary for curated list of code CAD

Irev-Dev, 2020-10-31:
Hey Matthew 👋 ,

I finally got around to putting together a list of all of the code-CAD projects that I know of [here](https://kurthutten.com/blog/curated-code-cad), and also on [Github](https://github.com/Irev-Dev/curated-code-cad).

I've put a short summary of each one, but it's hard to do them all justice. for libfive I've only got:
libfive is a software library and set of tools for solid modelling, especially suited for parametric and procedural design. lisp based language, (so (you (((((can expect ) lots of brakets))))).

If you would like to give me a summary of the project, and perhaps some points of how it differs from other similar projects would be happy to include it. I think max three paragraphs would be about right. Open a PR on reply to this issue and I'll include it.

Irev-Dev, 2021-01-01:
Me again.

I also want to add some of the best real-world examples of Code-CAD being used out in the wild. Do you know of any good projects using libfive that I can feature?

So far what I've found is:
- [Acedemics using it for reactor design](https://github.com/ukaea/paramak)
- [Lego compatible blocks](https://fxbricks.com/)
- [specialised in-house equipment for drones](https://twitter.com/robsmallshire/status/1343477584458346497)
- [Pruse 3d printer](https://github.com/prusa3d/Original-Prusa-i3)

smilberger, 2021-01-01:
Thanks for your work on spreading the word about these projects, @Irev-Dev !

Here's one real-world project Matt Keeter did with Libfive:
https://www.mattkeeter.com/blog/2017-09-15-iron/

Done in Ao, the predecessor of Libfive, but interesting and relevant to Libfive:
https://chriskohlhepp.wordpress.com/economics-robotics/full-3d-metal-printing-at-home/
https://chriskohlhepp.wordpress.com/economics-robotics/digital-micromanufacture/

I would really like to see some advocacy of projects like Libfive on sites like ifixit.com.  As Matt showed in his article, this tech can reduce junk going to landfills.  My own article is going to have to wait until I get a CNC machine.