libfive/libfive #364: Questions

giannissc, 2020-12-07:
Can this library output STEP/IGES/DXF  and .gcode files. Basically my question is how can the implicit representation used in libfive be converted to an explicit representation for CNC, lazer cutting and 3D printing.

giannissc, 2020-12-07:
Follow up question: Can implicit modelling be used alongside a geometric constraint solver?

TyOverby, 2020-12-07:
> how can the implicit representation used in libfive be converted to an explicit representation for CNC, lazer cutting and 3D printing.

This library can output `.stl` files, which you should then feed into a slicer / planner for CNC/laser/3d-printing.  This is ok because frequently, the tools that you use for generating gcode are very specific to the machine that you run them on.

arcograph, 2020-12-10:
@giannissc 
> Follow up question: Can implicit modelling be used alongside a geometric constraint solver?

There is some discussion about using constraints with implicit modeling in this HN post:

https://news.ycombinator.com/item?id=9248174

Determining features like edges of a cube for use with a constraint system are more difficult with implicit modeling (the biggest hint about this is the word "implicit"), just given a SDF representation of a shape.  Even automatically determining the bounding box has been an issue for Libfive.

My own thinking has moved towards doing a hybrid approach, having a more abstract shape object that manages a boundary box and features (like edges) for constraint modeling, then using SDF rendering as one back-end.