libfive/libfive #365: Rust binding?

virtualritz, 2021-01-09:
I am very interested in a Rust binding for this.
Mainly to add another toolbox of goodness to my 3D creative coding/generative art endeavors in that language.
I would have a stab at this myself but before I do, two questions:

1. Is anyone working on a Rust binding of this already?
2. @mkeeter  Would you start from the C API or the C++ API if you were to write a Rust binding for this lib?

mkeeter, 2021-01-09:
> 1. Is anyone working on a Rust binding of this already?

Not that I know of!

> 2. Would you start from the C API or the C++ API if you were to write a Rust binding for this lib?

I haven't checked on the state of `bindgen` in a while.  If it's powerful enough to wrap the full C++ API, I'd suggest using it (since that exposes the full library functionality).  Otherwise, the C API should be pretty easy to wrap, and it exposes enough API for common tasks.

virtualritz, 2021-01-09:
Supposedly `bindgen` got a lot better at wrapping C++ – with [these caveats](https://rust-lang.github.io/rust-bindgen/cpp.html#unsupported-features).
Does `libfive` rely on any of those unsupported features? If it doesn't I'll try auto-wrapping the C++ API tonight to see what I get.

mkeeter, 2021-01-09:
Hmm, some of the meshing gets tricky with templates, but I'm not sure how much is exposed in the headers – I'd say to give it a try and see how it goes!

smilberger, 2021-01-09:
The C API in Libfive itself wraps the C++ code, and in doing so adds some inefficiency, including allocating and copying rendered mesh data into C structs.

That said, if I was adding a binding, unless I needed the highest efficiency or some feature not in the C API, I probably would stick with C. I prefer the simplest approach to make something work, and the C API to Libfive *is* very straightforward and works without much fuss. My 2 cents...

If you do try binding with C++ and Rust, it would be interesting to hear your results. It would also be of interest to know what can be done with the C++ interface that can't be done with the C API.

mkeeter, 2021-01-09:
> If you do try binding with C++ and Rust, it would be interesting to hear your results. It would also be of interest to know what can be done with the C++ interface that can't be done with the C API.

Furthermore, if there are specific functions that you're missing, then it's totally reasonable to expand the C API.  So far, it's just been based on "what do the Scheme bindings need?".

smilberger, 2021-01-09:
> So far, it's just been based on "what do the Scheme bindings need?".

Except for maybe:

https://github.com/libfive/libfive/commit/e05fb6e65d96bfe1c65ab58c5e17de9008e00e78

which added libfive_mesh_coords(), etc., and was helpful in avoiding rebuilding mesh structures at a high level when I was working on the Common Lisp and Python bindings.

I noticed that in the Unity/C# binding of Libfive, the author apparently copied (marshalled) "native" C data into a newly allocated, managed C# array after rendering. Who knows, there may even be another copy or two before sending the mesh data to the GPU from Unity.

Unless doing a "proof of concept" binding (a beneficial thing), it does make sense to know what is going on in detail at the various levels of an application--no matter which API you bind to.

smilberger, 2021-01-14:
> It does make sense to know what is going on in detail
I'm working on a different approach for my own bindings, so taking my own advice, I've been reading the code more, especially the C++ API and looking back through some of the relevant binding issues...

> Is anyone working on a Rust binding of this already?
Another person did some work:

https://github.com/libfive/libfive/issues/288 

and produced a PR for the C API:

https://github.com/libfive/libfive/pull/292

virtualritz, 2021-01-14:
@mkeeter What are `libfive`s non-negotiable deps?

It looks like the build instructions in the README include all the deps for `studio` as well?

Maybe that could be split in the build instructions?

mkeeter, 2021-01-14:
Non-negotiable deps are everything except Qt and Guile:
> Qt and Guile are optional; if they aren't present, then the Guile bindings and Studio will not be included in the build (and cmake will print a message to that effect).

virtualritz, 2021-01-14:
Cheers! Maybe that sentence could be added to the build instructions?

mkeeter, 2021-01-14:
[That's where I copied it from!](https://github.com/libfive/libfive/blob/master/README.md#dependencies)

virtualritz, 2021-01-17:
I see. It wasn't clear to me at all. I think it would be clearer to add this in brackets to the deps themselves. Would you accept a PR for that? :)

mkeeter, 2021-01-17:
Sure, go for it!

virtualritz, 2021-01-17:
[Done](https://github.com/libfive/libfive/pull/367) (I hope).

virtualritz, 2021-01-18:
My `build.rs` segfaults. From the looks of it, during `bindgen`.
I.e. `libfive` gets built just fine via the `cmake` crate and then ... boom:
```
error: failed to run custom build command for `libfive-sys v0.1.0 (/Users/moritz/code/crates/libfive-sys)`

Caused by:
  process didn't exit successfully: `/Users/moritz/code/crates/libfive-sys/target/debug/build/libfive-sys-83fb0ebcef2353cf/build-script-build` (signal: 11, SIGSEGV: invalid memory reference)
 ```
Other people seem to have ran into [something like this](https://github.com/rust-lang/rust-bindgen/issues/493) too.
Repo [is here](https://github.com/virtualritz/libfive-sys).

virtualritz, 2021-05-05:
So I abandoned my tries to get the C++ header through `bindgen` for the time being.

I have a rough version of the Rust API in place – https://github.com/virtualritz/libfive-rs.

This contains `libfive` itself as a subrepo under `libfive-sys`. This is built from source. Currently this tracks my fork as I have added a `libfivestd_` prefix to the `stdlib.h` functions (see #432).

Feedback welcome. Probably best to build the docs to get an idea.
```
cargo doc -p libfive --no-deps --open
```

Not sure how to ever get this published on `docs.rs` since that is sandboxed so all dependencies must be part of the package.

With `boost` being one that is pretty much impossible due to `crates.io` size constraints, I guess.

virtualritz, 2021-05-08:
So I now have two implementations for `Tree` in Rust.

Using references with (mandatory) lifetimes:

```rust
pub struct Tree1<'a> {
    tree: sys::libfive_tree,
    _marker: PhantomData<*mut &'a ()>,
}

impl Tree<'static> {
    pub fn x() -> Self {
        Self {
            tree: unsafe { sys::libfive_tree_x() },
            _marker: PhantomData,
        }
    }

    ...
}

impl<'a> Tree<'a> {
    pub fn union_self<'b>(&'b self, b: &'b Tree) -> Tree<'b> {
        Tree {
            tree: unsafe { sys::libfivestd_union(self.tree, b.tree) },
            _marker: PhantomData,
        }
    }
    
    ...
}

impl<'a> Drop for Tree1<'a> {
    fn drop(&mut self) {
        unsafe { sys::libfive_tree_delete(self.tree) };
    }
}
```
This variant is straightforward. Everything takes references; including any operators. This is reasonably easy to implement. Leaf nodes have `'static` lifetimes. Everything else is tracked with individual lifetimes.

This all works. But everything being references, issues arise once you want to move `Tree1`s around because of Rust's move semantics. I.e. great for quick results and demo apps but probably not so much for a real CAD app.

The 2nd variant uses `Rc` and basically creates a copy of the tree to track ownership for getting deallocations right. Also this allows to easily keep copies of (sub)`Tree2`s and move them around in an app:

```rust
pub struct Tree2 {
    tree: Rc<sys::libfive_tree>,
    child_a: Option<Rc<Tree2>>,
    child_b: Option<Rc<Tree2>>,
}

impl Tree2 {
    pub fn x() -> Self {
        Self {
            tree: Rc::new(unsafe { sys::libfive_tree_x() }),
            child_a: None,
            child_b: None,
        }
    }
    
    pub fn union(self, b: Tree2) -> Self {
        Self {
            tree: Rc::new(unsafe { sys::libfivestd_union(*self.tree, *b.tree) }),
            child_a: Some(Rc::new(self)),
            child_b: Some(Rc::new(b)),
        }
    }
    
    ...
}

impl Clone for Tree2 {
    fn clone(&self) -> Self {
        Self {
            tree: Rc::new(*self.tree.as_ref()),
            child_a: self.child_a.as_ref().map(|t| t.clone()),
            child_b: self.child_b.as_ref().map(|t| t.clone()),
        }
    }
}

impl Drop for Tree2 {
    fn drop(&mut self) {
        if 1 == Rc::strong_count(&self.tree) {
            unsafe { sys::libfive_tree_delete(*self.tree) };
        }
    }
}
```

The second variant is modeled so that a tree has full ownership of all children. Subtrees can be cloned and deallocation happens as expected because of the use of `Rc`.

The issue here is the standard lib. Many operators take more than two operands. So if I want to wrap that I have two options:
1. Swap out `child_a` & `child_b` for a `children: Vec<Tree2>` and use the `stdlib.h` header.
2. Just RRIR, `stdlib_impl.cpp` that is.

I'm currently leaning towards 2. as that is probably also just 1h of search & replace, mostly. 

Any thoughts/suggestions/ideas?

virtualritz, 2021-05-09:
That sounds like a very neat option that saves me lots of code! Thanks heaps.

virtualritz, 2021-05-09:
Wouldn't this more ergonomically expressed like this though:

```cpp
libfive_tree libfive_tree_clone(libfive_tree ptr)
{
    libfive_tree clone = ptr;
    Tree(clone).release()
    return clone;
}
```

virtualritz, 2021-05-09:
I got everything working now. One thing I have done so far is use functional style since that is more idiomatic (i.e. `5.0.sin()` is what `sin(5.0)` looks like in Rust):

To do this my `gen_rust.py` just renames the 1st argument to `self`.
However, anything that is in the `shapes` or `generator` module (the latter includes all `array*` and the `extrude_z` ops), will be a static method:

```rust
use libfive::*;

let csg_shape = Tree::sphere(Tree::from(1.0), TreeVec3::default())
    .difference_multi(vec![
        Tree::sphere(0.6.into(), TreeVec3::default()),
        Tree::cylinder_z(
            0.6.into(),
            2.0.into(),
            TreeVec3::new(0.0, 0.0, -1.0),
        ),
        Tree::cylinder_z(
            0.6.into(),
            2.0.into(),
            TreeVec3::new(0.0, 0.0, -1.0),
        )
        .reflect_xz(),
        Tree::cylinder_z(
            0.6.into(),
            2.0.into(),
            TreeVec3::new(0.0, 0.0, -1.0),
        )
        .reflect_yz(),
    ]);

csg_shape.to_stl(
    &Region3::new(-2.0, 2.0, -2.0, 2.0, -2.0, 2.0),
    10.0,
    "csg_shape.stl",
)
.expect("Could not write STL file.");
```

virtualritz, 2021-06-02:
I just published [this on crates.io](https://crates.io/crates/libfive).