libfive/libfive #366: Getting `fold-shape` to return separate values?

omgitsraven, 2021-01-17:
I'm afraid my understanding of `values` and `values-from` is letting me down again, but I can't understand where I'm going wrong...

I'd like to make several slightly-different objects, so I'm using `fold-shape`. The issue is, in order to avoid the fuzzy overlapping edges from `union`ing those shapes, it would be nice if I could return them as separate primitives (and worry about the watertightness after export)...

Here's an example (with the mushy-edges greatly exaggerated just to demonstrate what I mean): 

```
(set-bounds! [-1 -1 -1] [5 5 1])
(set-quality! 11)
(set-resolution! 3)

(sphere 0.75 #[0 0 0])
(sphere 0.75 #[1 0 0])
(sphere 0.75 #[2 0 0])
(sphere 0.75 #[3 0 0])
(sphere 0.75 #[4 0 0])

(fold-shape (emptiness) (lambda (shape i)
  (values shape (sphere 0.75 #[i 2 0]))
) (iota 5))

(fold-shape (emptiness) (lambda (shape i)
  (union shape (sphere 0.75 #[i 4 0]) )
) (iota 5))
```

- The first row gives the results that I want, but I'd like not to have to spell them out manually.
- The second row is my best guess at how to achieve those results through fold-shape, but mysteriously, it gives *nothing*.
- The third row demonstrates that my general idea with fold-shape is sound, if I simply union the shapes instead.

Is there any way to use `fold-shape` this way? I've tried wrapping every part in every combination of `values` and `values-from` that I can think of, but I can't understand why it isn't working.

Admittedly: I can just use `union` and crank up the resolution; this isn't a show-stopper. But if the answer is something simple, I'd like to understand more about how this all works; I assume the problem has something to do with the `shape` values getting increasingly nested, but when I tried this:

```
(fold-shape (values (emptiness)) (lambda (shape i)
  (values (values-from shape) (sphere 0.75 #[i 2 0]))
) (iota 5))
```

the result was still nothing...

smilberger, 2021-01-17:
One way:

```
(set-bounds! [-1 -1 -1] [5 5 1])
(set-quality! 11)
(set-resolution! 10)

(apply values
  (fold-shape '()
              (lambda (shape i) 
                 (append shape (list (sphere 0.75 #[i 2 0]))))
              (iota 5)))
```

The ```fold-shape``` part constructs a list of shapes, which is then supplied as positional arguments to ```values``` using ```apply```.

omgitsraven, 2021-01-17:
Thank you, that works for the example I gave!

However, in trying to implement it, I realised that my example didn't actually capture the entire situation I was in after all; it stops working when it involves a function that returns several shapes...

The good news is, I found out how to work around it; the bad news is, that workaround is incompatible with how everything else in libfive usually works, in such a way that I feel like there's probably a compatible solution that I'm missing...

Here's a new example:
```
(set-bounds! [-1 -1 -2] [5 3 1.5])
(set-quality! 11)
(set-resolution! 3)

(define (severalShapesValues i row)
  (values (sphere 0.75 #[i (* 2 row) 0]) (cylinder 0.75 1 #[i (* 2 row) -1.5]) )
)

(severalShapesValues 0 0)
(severalShapesValues 1 0)
(severalShapesValues 2 0)
(severalShapesValues 3 0)
(severalShapesValues 4 0)

(define (severalShapesList i row)
  (list (sphere 0.75 #[i (* 2 row) 0]) (cylinder 0.75 1 #[i (* 2 row) -1.5]) )
)

(apply values
  (fold-shape '() (lambda (shape i)
    (append
      shape
      (severalShapesList i 1)
    )
  ) (iota 5) )
)
```

The only difference between `severalShapesList` and `severalShapesValues` is that they wrap their two shapes in either `list` or `values` before returning it, but right now, calling the wrong one from either place won't work... Is there any way to make a `severalShapes` function that can be used in *either* scenario, rather than needing them to return different things depending on where they're being used?

smilberger, 2021-01-17:
> calling the wrong one from either place won't work

I might be missing something here, but what exactly are you using multiple values (MVs) for?  My understanding was that, in Studio, they are primarily used to specify multiple independent shapes when returned to the top level.  If you don't have some other specific use for MVs, you might consider consistently using lists, and converting them to a MV return at the very end, like in my example.

MVs in Scheme are really an optimization for efficiently returning several conceptually different objects (for example using the stack) and not really a "proper" sequence like a vector or a list.  The caller and receiver of a function have to arrange to transfer them, or only the first one is used, which I think is what was going wrong in your original code.  By transforming back and forth from lists to values, you probably lose any efficiency gain anyway.

https://www.gnu.org/software/guile/manual/html_node/Multiple-Values.html

MV returns *are* useful in cases like integer division, which can return both parts of the result efficiently--and ignore the remainder by default.

omgitsraven, 2021-01-17:
I'm almost certainly the one who's missing something here; I've never written scheme outside of libfive, and I still don't entirely understand what either `list` or `values` do; most of your last reply went over my head I'm afraid.

Don't worry about it; I was hoping this would have a relatively simple answer that would demonstrate to me what's going on, but I think I need to brush up more on Scheme itself first. Thanks anyway!

closed 2021-01-17 by omgitsraven

smilberger, 2021-01-17:
I looked at your code again, and it seems like the idea is that you want to call functions like ```severalShapesValues``` directly to have Studio render them (as separate shapes).  But you also want to use them as part of a higher-level function, like ```fold-shape``` for multiple shapes.  Makes more sense to me now.

It seems like what you are doing really is getting around *rendering* issues by not using union.  The *modeling* of Libfive consistently represents a single shape as a function.  You are kind of doing "ad-hoc" modeling of multiple shapes using the renderer's multiple value feature outside of what Libfive currently offers at the modeling level--it doesn't really fit in.  Think of the functions you are writing here as collecting shapes for rendering, not modeling itself.

If you read more, have a look at Scheme lists, functions that operate on them, and higher level functions especially.  All of the technical stuff I said earlier really just means multiple values are a pain to use unless you have a specific reason--so use lists to group shapes until the very end.