libfive/libfive #371: building: says Guile is missing when it's not?

omgitsraven, 2021-01-23:
I'm trying to build libfive again for the first time in a while, and it's not letting me build Studio, complaining that I don't have the necessary version of Guile, but from what I can tell I do?

```
hazel@hazel-VirtualBox:~/stuff/libfive/libfive/build$ cmake -DCMAKE_PREFIX_PATH=/opt/Qt/5.14.0/gcc_64 ..
-- Checking for module 'guile-2.2>=2.2.1'
--   No package 'guile-2.2' found
-- Checking for module 'guile-3.0>=3.0.0'
--   No package 'guile-3.0' found
Checking dependencies:
  libfive:		✓
  libfive-guile:	✘   (needs Guile 2.2 or later)
  Studio:		✘   (needs Guile 2.2 or later)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hazel/stuff/libfive/libfive/build

hazel@hazel-VirtualBox:~/stuff/libfive/libfive/build$ sudo apt-get install guile-2.2
[sudo] password for hazel: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
guile-2.2 is already the newest version (2.2.3+1-3ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

hazel@hazel-VirtualBox:~/stuff/libfive/libfive/build$ sudo apt-get install guile-2.2-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
guile-2.2-dev is already the newest version (2.2.3+1-3ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

hazel@hazel-VirtualBox:~/stuff/libfive/libfive/build$ guile
GNU Guile 2.2.3
Copyright (C) 1995-2017 Free Software Foundation, Inc.

Guile comes with ABSOLUTELY NO WARRANTY; for details type `,show w'.
This program is free software, and you are welcome to redistribute it
under certain conditions; type `,show c' for details.

Enter `,help' for help.
scheme@(guile-user)> 
```

mkeeter, 2021-01-23:
Weird!

CMake should be using `pkg-config` to find Guile.  Can you run `pkg-config --list-all|grep guile` and see whether it finds anything?

On my machine, I see
```
$ pkg-config --list-all|grep guile
guile-3.0                           GNU Guile - GNU's Ubiquitous Intelligent Language for Extension
```

omgitsraven, 2021-01-23:
Just gives me another prompt immediately, no response!

mkeeter, 2021-01-23:
Okay, that indicates that `pkg-config` isn't finding it.

It looks like Ubuntu installs `guile-2.2.pc` (the file that `pkg-config` searches for) in `/usr/lib/x86_64-linux-gnu/pkgconfig/guile-2.2.pc` (according to [this search](https://packages.ubuntu.com/search?suite=focal&arch=amd64&mode=filename&searchon=contents&keywords=guile%20.pc))

If that file is present, then maybe tweak `PKG_CONFIG_PATH` to include `/usr/lib/x86_64-linux-gnu/pkgconfig`.  Oherwise, I'm not sure – I'd expect the package manager to put the `.pc` files in a place `pkg-config` can find them.

omgitsraven, 2021-01-23:
OK, yep; adding this `export` line to my build .sh worked:
```
cd ~/stuff/libfive/libfive/build
export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig
cmake -DCMAKE_PREFIX_PATH=/opt/Qt/5.14.0/gcc_64 ..
make -j4
```
Thanks!

closed 2021-01-23 by omgitsraven