libfive/libfive #372: Studio just gives a black window

omgitsraven, 2021-01-23:
### Description

When I launch Studio, I get a black window.
When I hover over the left half, I see the text-insertion cursor, and when I hover over the right half, I see the arrow cursor, so clearly SOMETHING is running, but I can't see it.
When I click on the right half, the window closes (presumably crashed). No error message.

### Versions

- Operating system: Ubuntu 18.04.5 LTS
- `libfive` commit hash: 4c22c3689cc79276d50976543c580d77ae101c01
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.14.0

### Additional Information

I'm running in a VM, so possibly my graphics drivers are doing something funky (although a much older build of Studio worked fine yesterday).

mkeeter, 2021-01-23:
Definitely sounds like a GPU driver issue – can you try running other OpenGL programs and seeing what they do?

Studio targets OpenGL 3.2, so you should also check whether the VM nominally supports that version (for a long time, Parallels was stuck at the venerable OpenGL 2.1).

omgitsraven, 2021-01-23:
Oh dear; `glmark2` displays all its demos fine, but it suggests that I have `GL_VERSION:    3.1 Mesa 20.0.8`...

I'll try and see if there's any way to get VirtualBox to support a newer version, but if not, do you remember when the bump to requiring 3.2 started, so I can downgrade again?

omgitsraven, 2021-01-23:
(and lastly, as a long shot: there's no chance that the dependencies that were stopping Studio from running on Windows have changed at all, have they?)

mkeeter, 2021-01-23:
I think that it was targeting OpenGL 3.3 to start, then was downgraded to 3.2 in PR #116 .  Maybe you can use `git bisect` from an arbitrarily old revision (e.g. b03e01af31bfe74a588e3e1d665659d9dadef365) and see where it works / breaks?

I haven't tried building Guile on Windows in a few years – that was the main dependency issue, as all the other libraries are cross-platform.

omgitsraven, 2021-01-23:
Oh wow, if it's only ever been *higher* than 3.1 then I guess it's not a version thing? I'll try bisecting soon...

and, the most recent posts I can find online about building Guile on Windows are from you, so, I guess I give up on that ;)

smilberger, 2021-01-23:
I'm on Ubuntu 18.04.1, x86_64 on-the-metal, Libfive commit 4c22c36, OpenGL 4.5

I got Studio with commit 4c22c36 working on my system. But I had a segfault before it showed anything after installing and running Studio. To fix it I manually uninstalled everything related to Libfive, including the shared library, Studio executable, and header files (it doesn't look like Libfive has a "make uninstall" command), then reinstalled the newest one. I don't know why, but that worked.

Git bisect might take a while on a VM, I thought you might try the above first.

omgitsraven, 2021-01-23:
By "manually uninstalling everything" do you just mean the `libfive` folder that was checked out from github, or also things like all the dependencies?

I threw out my entire libfive folder (including the build folder) and downloaded it fresh before trying to build again; I've definitely run into that problem in the past too.

omgitsraven, 2021-01-23:
Oh hey, I just realised that Studio outputs stuff to the terminal if you launch it from the terminal; does this narrow it down at all?
```
hazel@hazel-VirtualBox:~/stuff/libfive/libfive/build/studio$ ./Studio
libGL error: MESA-LOADER: failed to open swrast (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
libGL error: failed to load driver: swrast
QOpenGLWidget: Failed to create context
QOpenGLWidget: Failed to create context
qt.qpa.backingstore: composeAndFlush: QOpenGLContext creation failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
qt.qpa.backingstore: composeAndFlush: makeCurrent() failed
```

smilberger, 2021-01-23:
> By "manually uninstalling everything" do you just mean the libfive folder that was checked out from github, or also things like all the dependencies?

Sorry that wasn't clear...I meant I uninstalled the files that the previous Libfive "make install" had written, like ```/usr/local/include/libfive```, etc., *not* any of the dependencies. After doing so, I also deleted the build for the newest Libfive and rebuilt it. I thought something might be getting cross-versioned.

omgitsraven, 2021-01-23:
My `/usr/local/include` directory is empty; how can I find out where the makefile is writing to? (I don't know too much about working with Make...)

omgitsraven, 2021-01-23:
OK, this isn't great; b03e01af31bfe74a588e3e1d665659d9dadef365 doesn't actually run any more either, and neither does the formerly-working build that I pulled out of my Trash?

I'm starting to wonder if an Ubuntu is what broke it across the board or something? I remember installing one before shutting down recently :S

smilberger, 2021-01-23:
> how can I find out where the makefile is writing to? 

```<build-directory>/install_manifest.txt``` has a list of the installed files.

The terminal commands:

```sudo updatedb```
```locate libfive```

should give you a list every file containing ```libfive``` actually on your file system.

omgitsraven, 2021-01-23:
I don't see any `install_manifest.txt` file in the build directory; `locate libfive` works, but it only shows me files inside my checked-out directory (and the copies of it in the trash)...

smilberger, 2021-01-23:
In this state of your system, I think if you delete the build directory, mkdir a new one and run cmake, make, and install as usual, the cross-versioning thing shouldn't be a problem.  Does this work?

If you close your VM and restart it from the saved VM file, is the old Libfive pre-installed?  Does the older Libfive still run?

omgitsraven, 2021-01-23:
Yep, I deleted the entire libfive folder and downloaded a fresh one from github before starting the thread.

I found the previous libfive in my trash, and it no longer runs, so it might be some sort of system update that broke it,

omgitsraven, 2021-02-07:
With help from someone on twitter I got it working again!!

They suggested that the issue might be with Qt, specifically that by supplying the `-DCMAKE_PREFIX_PATH=/opt/Qt/5.14.0/gcc_64` flag, I wasn't using Ubuntu's own version of it... and I said that I was never able to find the path to Ubuntu's own version of it, but the idea made me think that maybe since it was built in, I wouldn't need to override any path... so I took that part out of the build script (making the `cmake -DCMAKE_PREFIX_PATH=/opt/Qt/5.14.0/gcc_64 ..` command into just `cmake ..`), and it built successfully and launched successfully!!

I'll still never understand why it suddenly broke, but at least it's over.

closed 2021-02-07 by omgitsraven

mkeeter, 2021-02-07:
Oh good, I'm glad that you got it working again!

It's frustrating how much software breaks if you just... leave it alone for a few months.