libfive/libfive #374: Video Example

SGSSGene, 2021-01-29:
Hey,
I have seen your videos of the examples: https://libfive.com/examples/
they look really cool. I would like to reproduce them. How did you create them?

SGSSGene, 2021-01-30:
I have integrated on a local branch into libfive-studio. Not sure if I am satisfied:
[Video of gears](https://raw.githubusercontent.com/SGSSGene/libfive/data/doc/gears.webm)

mkeeter, 2021-01-30:
Yup, looks very similar to what I did.

Here's the patch from when I was recording the videos (several years out of date at this point):
```
commit 54cbac1baf6e8375602dd9dbdc96021836143526 (render-video)
Author: Matt Keeter <matt.j.keeter@gmail.com>
Date:   Fri Dec 15 17:57:18 2017 -0500

    Spin 3D model for website demo videos

diff --git a/gui/src/background.cpp b/gui/src/background.cpp
index 6989d483..866f1eb3 100644
--- a/gui/src/background.cpp
+++ b/gui/src/background.cpp
@@ -27,10 +27,10 @@ void Background::initializeGL()
     {
         // Data is arranged  x   y   z   r   g   b
         GLfloat data[] = {   /********************/
-                            -1, -1,  0.9,  COLOR(base03),
-                            -1,  1,  0.9,  COLOR(base02),
-                             1, -1,  0.9,  COLOR(base02),
-                             1,  1,  0.9,  COLOR(base02),
+                            -1, -1,  0.9,  1,1,1,//COLOR(base03),
+                            -1,  1,  0.9,  1,1,1,//COLOR(base02),
+                             1, -1,  0.9,  1,1,1,//COLOR(base02),
+                             1,  1,  0.9,  1,1,1,//COLOR(base02),
         };
         vbo.create();
         vbo.bind();
diff --git a/gui/src/view.cpp b/gui/src/view.cpp
index e6705f60..1135140d 100644
--- a/gui/src/view.cpp
+++ b/gui/src/view.cpp
@@ -43,6 +43,13 @@ View::View(QWidget* parent)
     pick_timer.setSingleShot(true);
     pick_timer.setInterval(250);
     connect(&pick_timer, &QTimer::timeout, this, &View::redrawPicker);
+
+    auto t = new QTimer;
+    connect(t, &QTimer::timeout, this, [&](){
+        camera.rotateIncremental(QPoint(1, 0));
+        this->update();
+        });
+    t->start(20);
 }

 View::~View()
 ```

SGSSGene, 2021-01-30:
Ah! cool! Thank you!