libfive/libfive #378: Polyhedra Library

smilberger, 2021-02-11:
I am porting [Kaleido](http://harel.org.il/zvi/kaleido/dual.html) by Dr. Zvi Har’El from Java to Common Lisp, and I was wondering if there would be interest in my porting it to Guile Scheme for Studio as part of the shape library.  I've got it working pretty well in CL, and it wouldn't be too hard to convert it further to Scheme, if you think it would be useful.  (Kaleido has a BSD-style license.)  There are some SDF demos of polyhedra on Shadertoy that appear to use a similar kaleidoscopic method for generation, but I couldn't decipher them enough to create a library.

Kaleido provides 80 uniform polyhedra, including Plato's and Archimedes' solids, etc.  I'm currently using half-spaces to render the solids from the faces.  I was originally trying to improve my turtle model from Tovero and also create a [Koalahedron](https://demonstrations.wolfram.com/PolyhedralKoalas/) that Stephen Wolfram's daughter created in Mathematica.

Here's mine:

[image]

TyOverby, 2021-02-12:
I'm definitely interested!  Actually, I'd be interested in seeing your common-lisp code even if you don't port it to scheme.

mkeeter, 2021-02-12:
That's an adorable koalahedron!

I'm about to start a revamp of the `libfive` standard library, with the goal of making it more easily wrapped by arbitrary other languages.  The plan is to implement all of the stdlib shapes _once_ in C++, expose a C header, then autogenerate FFI using each language's native binding strategy (e.g. `ctypes` for Python).

This will save us from needing to rewrite the shapes library in every higher-level language!

The downside, of course, is that the canonical implementation of shapes will need to be in C++.  If you'd be interested, I'll link the new `libfive-stdlib` docs once they emerge; otherwise, please link the CL code, and _someone_ will probably port it eventually (also, like @TyOverby, I'm curious to see what it looks like 😀)

smilberger, 2021-02-12:
If you want to see the Kaleido code in C, it is a module in Xscreensaver, [here](https://salsa.debian.org/debian/xscreensaver/-/blob/master/hacks/glx/polyhedra.c) and [here](https://salsa.debian.org/debian/xscreensaver/-/blob/master/hacks/glx/polyhedra-gl.c).

I used the Java version because it was more up to date, but it is essentially the same as the C version (and mine in CL).

Har’El published a [paper](http://www.harel.org.il/zvi/docs/uniform.pdf) on the details of Kaleido.

smilberger, 2021-02-12:
@mkeeter, I like your idea about pushing the Guile code into the C++ layer.  Would you start a new issue when you are ready to have people look at it?  I've had a couple ideas about pre-calculated bounding boxes and reference counting that might make sense to add while you are doing it.  I'm not sure having the canonical implementation of shapes in C++ is *all* downside though--many more people know C++ than Guile.  Coding in Lisp feels more natural to me, but I know many people think Lisp is weird. :)

As long as the model remains extensible (e.g. adding new shapes, transforms, etc.) in a wrapping language like Guile or CL, I think the C++ would work well.  I wrote a molecule viewer (after reading your raytracing article) using my Pentane project in C, and I had to port some of the Guile code to C.  Doing shape arithmetic was clunky in C, without generic functions or operator overloading, but it did also occur to me that the standard library now in Scheme could be implemented at a lower level, and that would be useful.

I'll hold off until you have more done on the new library, then maybe port Kaleido directly to C++.  I've been trying to find a paper by a group (Brown University?) that did a comprehensive SDF library for polyhedra (I think in C++), but it's been years since I read it, and I haven't been able to find it again.  That might be another option if I ever do find it.

smilberger, 2021-02-19:
My recent work with polyhedra and thinking about generating texture coordinates in Libfive made me think of an interesting video lecture series on MIT OCW which I highly recommend:

https://ocw.mit.edu/courses/electrical-engineering-and-computer-science/6-849-geometric-folding-algorithms-linkages-origami-polyhedra-fall-2012/class-and-lecture-videos/

Dr. Demaine has pioneered some of the work on the math of geometric folding. I'm not sure if there has been much cross-pollination between signed distance field rendering techniques and geometric folding, but it seems like it might be applicable to designing and building artifacts that we can specify in Libfive (sheet metal patterns, for example) with non-additive methods.  It also has applicability to finding ways to unfold polyhedra to flat patterns without edge overlap (for determining texture coordinates).   Geometric folding is also a very interesting topic in itself in the areas of paper origami, robot motion, and protein folding.

closed 2021-02-21 by smilberger

koalahedron, 2021-03-02:
Now that Matt is getting the standard library into C/C++, I thought using the C code version could be integrated fairly simply.  One fly in the ointment might be that Kaleido has sort of a BSD license, but not a standard one, the license is [here](https://github.com/koalahedron/polyhedra/blob/main/polyhedra.lisp).  It just requires an attribution in the code, but I don't know if that requirement is compatible with MPL 2.0.

> I'm definitely interested! Actually, I'd be interested in seeing your common-lisp code even if you don't port it to scheme.

While I'm here, here is my CL code:

https://github.com/koalahedron/polyhedra

I'm planning on adding some OpenGL code from XScreensave to visualize the polyhedrons.  In Libfive, I'm currently only able to render the convex ones.  I would need some algorithm to convert concave ones into convex sub-units to use the half-space technique.

stew-j, 2022-07-24:
My 82 year old mother made Koalahedron into a quilt for the craft museum at the University of Colorado, Boulder:

[image]

She's been a "Maker" for 60+ years.