libfive/libfive #381: Studio regenerates on whitespace changes

omgitsraven, 2021-02-14:
### Description

[Description of the issue]
Whenever I type anything in Studio, _significant or not_, it triggers a remesh. (This didn't used to happen.)

[Steps to reproduce]
1. Create a new document
2. Add `(sphere 1)`, and observe that the 3D view builds
3. Hit enter, and observe that the 3D view **re**builds just from that

(Maybe add a more complex shape than a sphere if your machine is fast enough that you can't see the rebuild...)

[What did you expect to happen?  What happened?  All the time?]
It used to be that I could actually make pretty significant refactors without anything rebuilding at all, because the resulting expression was the same after parsing (i.e., deleting `define`s that weren't being used, or replacing a literal value with an identical `let` variable)
This is the first time that I've used Studio since the whole thing with getting the Tree branch to build, so it seems likely that it's related to that...

[Are there any error messages printed to the console?]
Nope.

### Versions

- Operating system: Ubuntu 18.04.5
- `libfive` commit hash: 8b057cf18b62de949f57ad578d8c3e28fb676053
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.9.5

mkeeter, 2021-02-14:
Interesting, I'll check it out – I didn't *intentionally* change this, but have some suspicions...

mkeeter, 2021-02-14:
I believe this is fixed in 9c3e9b861825f8b1a8740da67ea206b8b8833107.

(Also, let me know if you notice more lag in the GUI after this change – it's doing extra computation in the GUI thread to deduplicate trees, but is probably Fast Enough™)

omgitsraven, 2021-02-14:
That seems perfect, thank you! (I haven't noticed any GUI sluggishness yet, but I'll keep it in mind...)

closed 2021-02-14 by omgitsraven