libfive/libfive #382: SDF Visualization

smilberger, 2021-02-16:
Some of the literature and tutorial websites on SDFs have a 2D visualization of a slice of an SDF.  For example, David Fogelman has this capability in his Python SDF project (marching cubes):

https://github.com/fogleman/sdf#visualizing-the-sdf

Is there a straighforward way to do this with Libfive, even if just a bitmap I can render from an outside program?

mkeeter, 2021-02-17:
Yes, with performance depending on how far into the weeds you want to go!

At the C API level, you can use [`libfive_tree_eval_f`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive.h#L261), which returns the distance at a given point.  Call this a bunch of times in a regular grid, then save that data using `libpng`, or as a [PPM](https://en.wikipedia.org/wiki/Netpbm#PPM_example), or just print it and paste it into Python.

You could do the same thing in Scheme using `(shape-eval s [x y z])`, which calls `libfive_tree_eval_f` under the hood.

If you want higher performance, the [C++ `ArrayEvaluator` class](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/eval_array.hpp#L87) lets you [`set`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/eval_array.hpp#L33) a bunch of points, then evaluates them using [`values`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/eval_array.hpp#L87) (which does SIMD evaluation).  This will also be faster because it only constructs the evaluator once (vs once per call to `libfive_tree_eval_f`), but is probably unnecessary unless you're rendering _huge_ shapes.

smilberger, 2021-02-17:
D'oh! Sometimes I can't see the forest for the trees.

    (shape-eval (sphere 1) [.5 .5 .5])
    Result: -0.1339746117591858

It might make an interesting C API addition to use the ArrayEvaluator and return a bitmap.  Also another motivation for having "annotation objects" in Studio.

closed 2021-02-21 by smilberger