libfive/libfive #384: [studio] Add support for Wavefront OBJ export

virtualritz, 2021-02-18:
See subject.

Preferably the mesh would contain as many quads as possible. This would make Studio a great tool in polygonal/subdivision surface modeling pipeline.

smilberger, 2021-02-18:
It seems like the practicality of doing this would depend on the meshing algorithm--using manifold DC method might be more straightforward, since  IIRC it uses quads interally vs. the iso-simplex which uses triangles.

virtualritz, 2021-02-18:
Yes, absolutely. DC seems to give the best meshes for this use case anyway.

virtualritz, 2021-02-18:
I.e. the docs could mention this but ultimately it would the user's decision.
That being said: is it possible to switch the meshing method vie Scheme or only through the menu?

smilberger, 2021-02-18:
I think everything is currently rendered to triangles anyway including manifold DC: [dc_mesher.cpp](libfive/src/render/brep/dc/dc_mesher.cpp#L136)

> it would the user's decision

Right, but if they choose iso-simplex, for example, they might not be able to choose to export quads.

I don't think it is possible to change the method via Scheme (although it might be a good idea to add that to the C API, if not, so other language bindings could do so too).

smilberger, 2021-02-18:
Also, the note that "quads" aren't strictly planar (within a reasonable epsilon such as 1e-6) as generated internally by the manifold DC method in Libfive, which I found by comparing the corner normals.  I guess this would depend somewhat on the resolution and quality settings.

virtualritz, 2021-02-18:
Non-planarity is a non-issue. Particularly if those meshes are used as base shapes for render time subdivision surfaces.

smilberger, 2021-02-19:
I've been looking at [this](https://en.wikipedia.org/wiki/Wavefront_.obj_file) as a reference.

Would it help to put vertex normals in the output file, even though they are optional? Since we have them in Libfive anyway... Tex coords (optional in OBJ) aren't currently available in Libfive, but I wonder if any of these meshing algs. have a convenient way to generate them.  

Also, since the file format looks free form plain text, it might be possible to hack together a quick prototype with quads using printf statements--say by limiting the render to one thread or interleaving vertex points/face indices.  Since Libfive is currently generating triangles exclusively, there would have to be a fairly significant change to the code to support quads, I think, even for a proof of concept.

omgitsraven, 2021-02-19:
Normals would be nice; plucking them out of the distance field gives much better results than attempts at smoothing neighbouring vertices in my experience, at least for low enough resolution models. (This isn't really a use case that I've had for libfive yet, but more options is always good.)

virtualritz, 2021-02-20:
The best reference is [this](http://paulbourke.net/dataformats/obj/). A copy of the original format description from Wavefront. 

Vertex normals are useful. They can be used to discover which edges to tag as [creases](https://graphics.pixar.com/opensubdiv/docs/subdivision_surfaces.html#semi-sharp-creases) and which vertices as corners, when rendering a mesh as a subdivision surface.

At the top Paul also links to a page that documents non-standard extensions that some apps are able to read and which may come handy, at some stage. I.e. if `libfive` ever adds support for attaching properties to SDFs and store them on the mesh. E.g. color.

smilberger, 2021-02-20:
> The best reference is

@virtualritz It is clearer, thanks!  WRT vertex normals, it nicely describes happens when they are included or omitted--I'm sure it varies by application IRL, though.  It looks like it might be a good idea to optionally include them, in any case.  More work of course, but pushing the quads through to the shader with/without these normals to directly visualize the result is potentially feasible in Studio--most likely the shader is tied to tris only, currently.

Re: properties, this could be another interesting improvement.  I think HyperFun has the concept of "parallel" (to the SDF) continous property "fields", like for 3D color properties, which you evaluate at points.

@omgitsraven By "we have them in Libfive already", I meant that [here](https://github.com/libfive/libfive/blob/c4c01f6a18af7bd6bc1d35a4fdbd3cb73618282d/libfive/src/render/brep/dc/dc_mesher.cpp#L142) for instance, the code is calculating the vertex normal in this helper lambda by normalizing a cross between edges--not that we are getting a smooth normal off the SDF.  That is lost due to quantization error anyway during meshing.  So long story short, I thought there might be some size/performance tradeoff.  The normals Libfive generates during meshing can be recovered later anyway if we know the edges and their ordering.

At some point, adding all this new stuff to the code could affect the real-time display performance--so some refactoring of what is currently done is probably in order.  However, if done right, it might make further support of other formats more straightforward.

smilberger, 2021-02-20:
An obvious separation of concerns that my previous post suggests is what Libfive needs to move forward (F-rep, I'm assuming), what it needs to provide to support interop with legacy technology (B-rep), and how should they work together.  The code already is organized along those lines.  Still, there is some ambiguity.  Since Studio renders to triangles to display using OpenGL, having wireframe would be useful while that is still the case.  But for direct (non-tesselating) techniques like sphere tracing, wireframing might be more suitable to use in an external tool like meshlab--examining the results of an exported mesh, for example.  Cooperatively supporting existing systems is great, but when we spend our limited time catering to them, we do lose other opportunities.  Animation can be done in F-rep too, meshlessly, as can engineering analysis.  That said, I don't think OBJ format would be that big of a modification to worry about, but it is non-trivial.

mkeeter, 2021-02-21:
Like @smilberger said, dual contouring uses quads internally, so this isn't too weird – just lots of plumbing.

Sprinkling `printf` statements into `dc_mesher.cpp` would definitely be a good way to hack this together!

A more serious implementation would add quads to [`per_thread_brep.hpp`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/render/brep/per_thread_brep.hpp), then push them through the rest of the pipeline.  It's awkward to do that while still making it arbitrary-dimension; I guess in N-Dimensional dual contouring, each primitive has 2^N vertices, and is split down into subshapes of size N, which happen to be equal in 2D.  The right abstraction is probably a `Branes<N, M=ipow(2,N)>`, which is statically unrolled until `M == N`.

On a related note – [in the past, I attempted](https://github.com/libfive/libfive/compare/smooth-rendering-again) to annotate the mesh data with whether each vertex was tagged as a face (smooth), edge (creased), or corner (sharp).  This ended up revealing deficiencies in the meshing which still haven't been fixed: in some cases, triangles are coplanar with opposite normals, which looks bad, so I abandoned it.

smilberger, 2021-02-21:
> Sprinkling printf statements into dc_mesher.cpp would definitely be a good way to hack this together!

I call "dibs"!  (I have given up resisting the "Big B-rep" lobbyists.  Also, Brep isn't obsolete just [yet](https://github.com/libfive/libfive/issues/378#issuecomment-782432478).  I also almost have it done.)
 
----

"Never volunteer" --every soldier ever

smilberger, 2021-02-21:
@mkeeter I thought I would use the Guile REPL to initiate a render, to "simplify" things like progressive rendering.  ("I almost have it done" is also almost always wrong.)  Anyway, I'm guessing this command from guide.md no longer applies due to your recent changes:

    (load-extension "libfive/bind/libfive-guile" "scm_init_libfive_modules")

Which gives me:

    scheme@(guile-user)> (load-extension "libfive/bind/libfive-guile" "scm_init_libfive_modules")                                 
    ice-9/boot-9.scm:1669:16: In procedure raise-exception:                                                                       
    In procedure dynamic-link: file: "libfive/bind/libfive-guile", message: "file not found"                                      

Any tips on fixing this would be appreciated.

smilberger, 2021-02-21:
Looks like that step isn't needed now, but I do have to figure out which modules to use
```
scheme@(guile-user)> (use-modules (libfive kernel) (libfive vec) (libfive csg) (libfive shapes))                              
While compiling expression:                                                                                                   
no code for module (libfive csg)                                                                                              
scheme@(guile-user)> (circle 1)                                                                                               
;;; <stdin>:7:0: warning: possibly unbound variable `circle'                                                                  
ice-9/boot-9.scm:1669:16: In procedure raise-exception:                                                                       
Unbound variable: circle                                                                                                      
```

mkeeter, 2021-02-22:
Good catch, this should be fixed in 8af59dcabbed2f9cee9e39d53cc9e565ddd7d55c!

smilberger, 2021-02-22:
I'm guessing this isn't how a sphere is supposed to look in OBJ, but it's a start.  Tris only for now.

[image]

smilberger, 2021-02-23:
This is using quads only in an OBJ file (in meshlab):

[image]

[image]

smilberger, 2021-02-23:
[My progress](https://github.com/smilberger/libfive/tree/obj-format-dev)