libfive/libfive #395: `sequence` equivalent for python?

omgitsraven, 2021-03-18:
I like using the `sequence` function (macro?) to keep arguments near the function name when making very very nested things... is there any possibility of making some similar syntax in Python?

mkeeter, 2021-03-18:
Have you discovered all of the member functions of the `Shape` class?  It's got CSG functions and the whole `libfive.transforms` namespace (inserted [here](https://github.com/libfive/libfive/blob/master/libfive/bind/python/libfive/shape.py#L162-L172)), so you can do something like

```python
sphere(1)\
    .scale_z(2)\
    .difference(cylinder(0.4, 10, [0, 0, -5]))
```

(the `\` are needed because otherwise Python thinks it's the end of the expression)

omgitsraven, 2021-03-18:
oh I absolutely had not!  That's perfect, thank you :)

closed 2021-03-18 by omgitsraven