libfive/libfive #400: Remarks and questions regarding python bindings

occivink, 2021-03-21:
First of all thanks for the python bindings, I previously tried to get into libfive but wrapping my head around both scheme and functional representation at once was a little difficult.

To get my feet wet, I've been re-implementing some primitives from [Inigo Quilez' webpage](https://iquilezles.org/www/articles/distfunctions/distfunctions.htm). I have some questions and remarks which came up as I went:

1. Some python shape bindings seem to be missing. I'm not sure if it's a simple omission, but I've used this patch to expose what I needed
<details>
  <summary>Patch</summary>
  
  ```diff
diff --git a/libfive/bind/python/libfive/shape.py b/libfive/bind/python/libfive/shape.py
index f2fc83ad..b4d5ec96 100644
--- a/libfive/bind/python/libfive/shape.py
+++ b/libfive/bind/python/libfive/shape.py
@@ -112,6 +112,9 @@ class Shape:
     def __truediv__(self, other):
         return Shape.new('div', self.ptr, other.ptr)
 
+    def __neg__(self):
+        return Shape.new('neg', self.ptr)
+
     def __eq__(self, other):
         raise RuntimeError("Shape does not support equality comparisons")
 
@@ -150,6 +153,20 @@ class Shape:
         return Shape.new('log', self.ptr)
     def square(self):
         return Shape.new('square', self.ptr)
+    def abs(self):
+        return Shape.new('abs', self.ptr)
+
+    @_wrapped
+    def min(self, other):
+        return Shape.new('min', self.ptr, other.ptr)
+
+    @_wrapped
+    def max(self, other):
+        return Shape.new('max', self.ptr, other.ptr)
+
+    @_wrapped
+    def compare(self, other):
+        return Shape.new('compare', self.ptr, other.ptr)
 
     def save_stl(self, filename, xyz_min=(-10,-10,-10), xyz_max=(10,10,10),
                  resolution=10):
  ```
</details>

2. The capped torus is where I ran into my first issue, with expressing conditionals such as in this line:
`float k = (sc.y*p.x>sc.x*p.y) ? dot(p.xy,sc) : length(p.xy);`
I've found a way using `compare` like so,
```
comp = Shape.max(Shape.compare(px * sc[1], py * sc[0]), 0)
k = comp * (px * sc[0] + py * sc[1]) + (Shape.wrap(1) - comp) * (px.square() + py.square()).sqrt()
```
but it feels roundabout. Is there a better approach?

3. Doing something like `1 + Shape.X()` doesn't automatically wrap 1. I have a feeling that this is an unsolvable python thing, though.

4. `remap` isn't currently exposed in python, is that right?

mkeeter, 2021-03-21:
> Some python shape bindings seem to be missing. I'm not sure if it's a simple omission, but I've used this patch to expose what I needed

Thanks, I added that patch!

> Doing something like 1 + Shape.X() doesn't automatically wrap 1. I have a feeling that this is an unsolvable python thing, though.

I implement `__r(add/sub/truediv/etc)__`, which fixes this.

> `remap` isn't currently exposed in python, is that right?

It is now!

> The capped torus is where I ran into my first issue, with expressing conditionals such as in this line:

Conditionals have always been a tricky subject: a lot of `libfive`'s meshing assumes C0 continuity (i.e. `f(xyz + epsilon) --> f(xyz) as epsilon --> 0`  This is very easy to break with conditionals, and is one of the few things that will segfault the kernel, so they always make me nervous.  In addition, the tree pruning / tape shortening algorithm doesn't handle them well, so they're not evaluated very efficiently.

At a brief glance, what you're doing is as fine as you can get with `libfive` right now, but if you could find a way to structure it without the conditional entirely, that would be more robust.

occivink, 2021-03-22:
Thanks for the clarifications and the fixes. The continuity requirement you mention is on the signed distance function, yes? If so, is something such as `Shape.compare(half_space([0,1,0]), sphere(1))` invalid? It does seem to behave properly in studio.

mkeeter, 2021-03-23:
Try to intersect it with something 😄 

For instance, `half_space([0,1,0]) - sphere(1)` has the same sign as `Shape.compare(half_space([0,1,0]), sphere(1))`, but is C0-continuous.  Look at what happens when you intersect both with a cylinder:

[image]

[image]