libfive/libfive #401: Errors cause bounds change and remesh, in Python

omgitsraven, 2021-03-21:
### Description

1) Enter the following file:
```
import studio
studio.set_bounds([-1, -1, -1], [1, 1, 1])
studio.set_quality(8)
studio.set_resolution(100)

from libfive.stdlib import *
sphere(0.9)
```

2) Select `View` > `Show bounding box(es)`

3) Select `View` > `Zoom to bounds`

4) Type a letter on the end of the python source to cause an error (i.e. change `sphere(0.9)` to `sphere(0.9)q`

You will immediately see the bounding box wireframe change size, AND see the mesh be remeshed into lower quality.

This isn't the end of the world, but it's very distracting because it happens any time I type anything in the file's source; it snaps to this low res version, before snapping back to how it should be as soon as I finish typing enough for the source to be valid Python again (and then snapping back to low-res again if I continue typing...)

### Versions

- Operating system: Win10 19042
- `libfive` commit hash: 0b9989d65192de05d04a8736fb42bbeab236d429
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): I'm actually not sure how to determine this on Windows; I guess `qt5-base`?

mkeeter, 2021-03-26:
Okay, this was an easy one – should be fixed now!

closed 2021-03-26 by mkeeter

omgitsraven, 2021-03-26:
Perfect, thanks!

commit b4e0e0bb (2021-03-26): Only change render settings if okay (fixes #401)