libfive/libfive #402: Allow multi-line indentation changes when editing python

omgitsraven, 2021-03-21:
### Description

If I've written some code, and I want to move it into a function so that I can reuse it elsewhere, Python requires that I change the indentation. However, if I select several lines and hit tab (or shift-tab), it just erases my selection.

This seemed to work in Scheme, so I'm not sure if it's a Scheme/Python difference, or a Linux/Windows difference, or a difference in my version of Qt...

### Versions

- Operating system: Win10 19042
- `libfive` commit hash: 0b9989d65192de05d04a8736fb42bbeab236d429
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): I would love to be more specific for this issue, since it seems like it might be relevant, but all the instructions that I can find online about how to find my Qt version only talk about Linux, so I might need some extra help...

mkeeter, 2021-03-21:
The reason this is different is because Guile has a [custom `Formatter`](https://github.com/libfive/libfive/blob/master/studio/src/guile/formatter.cpp) (which implements special behavior on certain keys), while [Python's `Formatter`](https://github.com/libfive/libfive/blob/master/studio/src/python/formatter.cpp) just ignores the key event.  It wouldn't be too hard to copy relevant code from one to the other!

mkeeter, 2021-03-23:
Some improvements in a5f2df2843387b35f40c02db2c26c9382f51b317, one more feature required (indentation matching in `Return`)

mkeeter, 2021-03-23:
Okay, this is now in decent shape – it's not too fancy, but works fine.

closed 2021-03-23 by mkeeter

omgitsraven, 2021-03-23:
Perfect, thank you so much :)