libfive/libfive #403: No `blend` on shape object in Python

omgitsraven, 2021-03-21:
### Description

This works:
```
blend(sphere(1),box_centered([4,0.5,0.5]),0.5)
```
This doesn't:
```
sphere(1).blend(box_centered([4,0.5,0.5]),0.5)
```

### Versions

- Operating system: Win10 19042
- `libfive` commit hash: 69b01f1e8d2edbc7db7af30be65f36ab4ff16358
  - Is this the latest commit to `master`?  yes

mkeeter, 2021-03-21:
This is another function that we should inject into the `Shape` class!

I think that the most general-purpose way to handle this is to add `mypy`-style annotations to `gen_py.py` and the Python `stdlib`, then add _any_ function whose first argument is a `Shape` (i.e. a `libfive_tree` in `stdlib.h`).

This would correctly add things like `array_x` to the `Shape` class, but wouldn't give it a `cylinder_z` member function (which would be nonsensical).

omgitsraven, 2021-03-21:
Actually, that made me realise one important difference between this and `sequence` -- new functions defined in a document won't be able to take advantage of it... (unless there's a way to monkey-patch new methods onto a class at runtime? I also know very little Python unfortunately, hahah...)

mkeeter, 2021-03-23:
Oh, that's a good point!

It would be pretty easy to implement a `Shape.call(fn, *args, **kwargs)`, e.g.
`circle(1).call(reflect_xz).call(move, [0, 0, 1])`, something like

```
class Shape:
    ...
    def call(self, fn, *args, **kwargs):
        ''' Calls the given function with self as the first argument.
            This is convenient to chain transforms without deep call trees.
        '''
        return fn(self, *args, **kwargs)
```

Does that seem reasonable?

omgitsraven, 2021-03-23:
Oh that actually sounds great as a workaround, sure! Good call, thank you :)

mkeeter, 2021-03-23:
Okay, added `Shape.call` in that commit!

omgitsraven, 2021-03-23:
That works, thanks so much!!

I'm not sure whether to close the issue now, or if I might as well leave it open until at least the stdlib functions are on Shape like you were originally considering? (This works fine as a workaround, though!)