libfive/libfive #404: Use fewer cores / throttling?

omgitsraven, 2021-03-21:
So, I was delighted to finally be able to throw my entire PC at libfive, rather than just however many cores I set aside for my VM... but in fact, I'm discovering that if I have (for example) a Twitch stream open in a browser, and I make some changes to an open libfive document, libfive will hit all of my cores so hard that the stream starts to stutter...

Is there any way to tell libfive to multitask a little more generously? (I'm not sure exactly what to ask for; I don't know particularly much about native development or how modern multitasking actually works...)

TyOverby, 2021-03-22:
what platform are you on?

omgitsraven, 2021-03-22:
Ah, sorry -- Windows 10

mkeeter, 2021-03-23:
This is a reasonable feature!

The dependency chain for this is:
- Persistent settings (using `QSettings`)
- A settings window (which includes a way to select number of rendering cores)
- Pass this through to [`Shape::renderMesh`](https://github.com/libfive/libfive/blob/master/studio/src/shape.cpp#L368-L380) (which calls `libfive::Mesh::render`)

Limeth, 2021-08-07:
I have the opposite issue when using Studio, it seems to only take advantage of 8 of my cores even though more are available.