libfive/libfive #408: Can't build on Ubuntu

omgitsraven, 2021-03-24:
### Description

Doing a fresh build gives this result:
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/gcc-7 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++-7 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /home/linuxbrew/.linuxbrew/lib/cmake/Boost-1.73.0/BoostConfig.cmake (found version "1.73.0")  
-- Found PkgConfig: /home/linuxbrew/.linuxbrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.3
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.7
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.34") 
-- Found Python3: /home/linuxbrew/.linuxbrew/opt/python@3.8/include/python3.8 (found version "3.8.5") found components: Development Development.Module Development.Embed 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
Checking dependencies:
  libfive:		✓
  Guile bindings:	✓
  Python bindings:	✓
  Studio:		✓   (Python + Guile)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hazel/stuff/libfive/libfive/build
Scanning dependencies of target libfive-python
Scanning dependencies of target libfive
[  1%] Generating libfive/runner.py
[  1%] Generating libfive/ffi.py
[  2%] Generating libfive/__init__.py
[  3%] Generating libfive/shape.py
[  3%] Generating libfive/stdlib/__init__.py
[  4%] Generating libfive/stdlib/csg.py
[  5%] Generating libfive/stdlib/text.py
[  5%] Generating libfive/stdlib/transforms.py
[  6%] Generating libfive/stdlib/shapes.py
[  6%] Built target libfive-python
[  6%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/deck.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_interval.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_jacobian.cpp.o
[  9%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_array.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_deriv_array.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_feature.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/tape.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/feature.cpp.o
[ 12%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/discrete/heightmap.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/discrete/voxels.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/contours.cpp.o
[ 14%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/edge_tables.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/manifold_tables.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/mesh.cpp.o
[ 16%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/neighbor_tables.cpp.o
[... 205 lines cut ...]
[ 96%] Building CXX object studio/CMakeFiles/Studio.dir/src/guile/language.cpp.o
[ 97%] Building CXX object studio/CMakeFiles/Studio.dir/src/guile/syntax.cpp.o
[ 97%] Building CXX object studio/CMakeFiles/Studio.dir/src/python/formatter.cpp.o
[ 98%] Building CXX object studio/CMakeFiles/Studio.dir/src/python/interpreter.cpp.o
[ 98%] Building CXX object studio/CMakeFiles/Studio.dir/src/python/language.cpp.o
[ 99%] Building CXX object studio/CMakeFiles/Studio.dir/src/python/syntax.cpp.o
[100%] Building CXX object studio/CMakeFiles/Studio.dir/Studio_autogen/EWIEGA46WW/qrc_resources.cpp.o
[100%] Linking CXX executable Studio
/home/linuxbrew/.linuxbrew/bin/ld: /home/linuxbrew/.linuxbrew/opt/python@3.8/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(dynload_shlib.o): undefined reference to symbol 'dlsym@@GLIBC_2.2.5'
/home/linuxbrew/.linuxbrew/bin/ld: /lib/x86_64-linux-gnu/libdl.so.2: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
studio/CMakeFiles/Studio.dir/build.make:559: recipe for target 'studio/Studio' failed
make[2]: *** [studio/Studio] Error 1
CMakeFiles/Makefile2:376: recipe for target 'studio/CMakeFiles/Studio.dir/all' failed
make[1]: *** [studio/CMakeFiles/Studio.dir/all] Error 2
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```

### Versions

- Operating system: Ubuntu 18.04.5 LTS
- `libfive` commit hash: 3d6f7c8f38ddc4fba6e04e532f30627642f52405
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.14.0

mkeeter, 2021-03-24:
Interesting – last time I checked, building everything works out of the box works on the latest Linux LTS (20.04).

A [bit of Googling](https://stackoverflow.com/questions/63153342/cmake-undefined-reference-to-symbol-dlsymglibc-2-2-5-even-though-i-link-with) suggests adding `-ldl` to the `target_link_libraries` call [here](https://github.com/libfive/libfive/blob/master/studio/CMakeLists.txt#L113).

(if this works, you should keep it as a local patch instead of committing it upstream, since I'm trying to avoid filling the CMakeFiles with support for a bunch of different Linux versions)

omgitsraven, 2021-03-24:
Replaced one error with another, which is a sort of progress!
```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/gcc-7 - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/g++-7 - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /home/linuxbrew/.linuxbrew/lib/cmake/Boost-1.73.0/BoostConfig.cmake (found version "1.73.0")  
-- Found PkgConfig: /home/linuxbrew/.linuxbrew/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.3
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.7
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.34") 
-- Found Python3: /home/linuxbrew/.linuxbrew/opt/python@3.8/include/python3.8 (found version "3.8.5") found components: Development Development.Module Development.Embed 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
Checking dependencies:
  libfive:		✓
  Guile bindings:	✓
  Python bindings:	✓
  Studio:		✓   (Python + Guile)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/hazel/stuff/libfive/libfive/build
Scanning dependencies of target libfive-python
Scanning dependencies of target libfive
[  1%] Generating libfive/__init__.py
[  2%] Generating libfive/ffi.py
[  2%] Generating libfive/runner.py
[  3%] Generating libfive/stdlib/__init__.py
[  3%] Generating libfive/shape.py
[  4%] Generating libfive/stdlib/csg.py
[  5%] Generating libfive/stdlib/shapes.py
[  6%] Generating libfive/stdlib/transforms.py
[  6%] Generating libfive/stdlib/text.py
[  6%] Built target libfive-python
[  6%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/deck.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_interval.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_jacobian.cpp.o
[  9%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_array.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_deriv_array.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_feature.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/tape.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/feature.cpp.o
[ 12%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/discrete/heightmap.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/discrete/voxels.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/contours.cpp.o
[ 14%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/edge_tables.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/manifold_tables.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/mesh.cpp.o
[ 16%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/neighbor_tables.cpp.o
[ 16%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/progress.cpp.o
[ 17%] Building CXX object libfive/src/CMakeFiles/libfive.dir/render/brep/dc/marching.cpp.o
[... 197 lines cut ...]
[ 94%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/voxels.cpp.o
[ 95%] Building CXX object studio/CMakeFiles/Studio.dir/src/python/syntax.cpp.o
[ 95%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/vol_tree.cpp.o
[ 96%] Building CXX object studio/CMakeFiles/Studio.dir/Studio_autogen/EWIEGA46WW/qrc_resources.cpp.o
[ 96%] Linking CXX executable Studio
[ 97%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/xtree.cpp.o
[ 98%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/util/mesh_checks.cpp.o
/home/linuxbrew/.linuxbrew/bin/ld: /home/linuxbrew/.linuxbrew/opt/python@3.8/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): in function `os_openpty':
posixmodule.c:(.text+0x66b4): undefined reference to `openpty'
/home/linuxbrew/.linuxbrew/bin/ld: /home/linuxbrew/.linuxbrew/opt/python@3.8/lib/python3.8/config-3.8-x86_64-linux-gnu/libpython3.8.a(posixmodule.o): in function `os_forkpty':
posixmodule.c:(.text+0xe4ed): undefined reference to `forkpty'
collect2: error: ld returned 1 exit status
studio/CMakeFiles/Studio.dir/build.make:559: recipe for target 'studio/Studio' failed
make[2]: *** [studio/Studio] Error 1
CMakeFiles/Makefile2:376: recipe for target 'studio/CMakeFiles/Studio.dir/all' failed
make[1]: *** [studio/CMakeFiles/Studio.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 98%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/util/shapes.cpp.o
[ 99%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/guile.cpp.o
[100%] Linking CXX executable libfive-test
[100%] Built target libfive-test
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2
```
TBH though, at this point I might just try starting fresh with a 20.04 VM instead...

mkeeter, 2021-03-24:
Since this is a VM, I agree that it would be easier to start from 20.04 – it looks like the Python 3 installation has _opinions_ on what it should be linked against, which I didn't encounter on 20.04.

mkeeter, 2021-03-27:
Fixed by building on 20.04 (#409)

closed 2021-03-27 by mkeeter