libfive/libfive #409: How to find Qt path on Ubuntu 20.04.2 LTS

omgitsraven, 2021-03-27:
This again!

Searching the drive for `gcc_64` didn't give any results this time, even though `qmake -v` gives
```
hazel@hazel-VB20:/usr/bin$ qmake -v
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
```

What exactly should I be looking for when determining the path to use for `-DCMAKE_PREFIX_PATH`? (As in, not just the name of the directory, but what is supposed to be *found in* that directory that I should be searching for?

For what it's worth, I tried building without any `-DCMAKE_PREFIX_PATH` value, and it seemed to complete successfully, but running Studio from the command line gives
```
hazel@hazel-VB20:~/stuff/libfive/libfive/build/studio$ ./Studio
guile: uncaught exception:
no code for module (libfive kernel)
```
I just mention it because I'm not sure if that's something that would be caused by my leaving out the prefix path, or if it might be an unrelated problem?

mkeeter, 2021-03-27:
If compilation builds Studio without `-DCMAKE_PREFIX_PATH`, then you're fine (and I can remove that from the Ubuntu 20.04 instructions).

I think the `no code for module` error is due to the path: try running it from `libfive/build` instead of `libfive/build/studio`, (this will require calling `./studio/Studio` instead).

omgitsraven, 2021-03-27:
That worked, thank you!! (Glad I checked!)

closed 2021-03-27 by omgitsraven

mkeeter, 2021-03-27:
Great, I'll update the README and close the previous issue about building on Ubuntu 18.