libfive/libfive #410: Studio.desktop not working on Ubuntu 20

omgitsraven, 2021-03-27:
I'd like to add Studio to my favourites bar, but given that it only works correctly when launched from a different directory, I'm not sure the best way to do this.

The Readme indicates a .desktop file that can be used, but even after adding it to `~/.local/share/applications` it isn't appearing in my Applications screen.

Very possibly related: I notice that the value for `Exec` in the .desktop file is just `Studio`, but running `Studio` on its own in the terminal just gives me `Studio: command not found`... (same with `studio`...)

omgitsraven, 2021-03-27:
Hold on, I just noticed I missed part of the instructions. One moment

omgitsraven, 2021-03-27:
OK, sorry about that, but: after running `sudo make install`, I'm now getting the same problem with a different error:
```
hazel@hazel-VB20:~$ Studio
Studio: error while loading shared libraries: libfive.so: cannot open shared object file: No such file or directory
```

omgitsraven, 2021-03-27:
However, I can now launch it from any directory as long as I put the entire path, i.e. (in my case)
```
hazel@hazel-VB20:~$ ~/stuff/libfive/libfive/build/studio/Studio
```

omgitsraven, 2021-03-27:
Also, sorry this is so scattershot, but: I just noticed that it IS appearing in my Applications menu now; it just doesn't do anything when I click it -- not even when I change the `Exec` value from `Studio` to `/home/hazel/stuff/libfive/libfive/build/studio/Studio`

mkeeter, 2021-03-27:
There might be multiple things going on here.

Let's debug the first one first, which is "Why does running `Studio` from the home directory fail after `sudo make install`?"

```
hazel@hazel-VB20:~$ Studio
Studio: error while loading shared libraries: libfive.so: cannot open shared object file: No such file or directory
```

I don't have a Linux system to test on, but `sudo make install` should be putting `Studio` into `/usr/local/bin` and libraries into `/usr/local/lib`.  You can find where `Studio` is installed by running `which Studio`; can you see whether `libfive.so` is nearby?

There's more discussion of shared libraries [here](https://stackoverflow.com/a/21173918).  The stuff about `$LD_LIBRARY_PATH` could be relevant: if `libfive.so` isn't in this path, then that would explain why it's not being detected.

omgitsraven, 2021-03-27:
OK this is totally baffling...  Without even rebooting, but AFTER having already relaunched the terminal and so on... `Studio` on the command line is now working, and so is the .desktop file. I have no idea what could have changed.

I almost hate to leave it at this, but: it's working, so never mind? Oh well...

closed 2021-03-27 by omgitsraven