libfive/libfive #411: Studio no longer actually opening files, on any platforms

omgitsraven, 2021-03-27:
### Description

[Description of the issue]
Going to File > Open will change the filename displayed in the window title, but will actually just reload the default file rather than the contents of the requested file.

[Steps to reproduce]
1. Launch Studio.
2. Change the contents of the default document (i.e. add a `box` to the scene)
2. Under the File menu, select Open
3. Pick a .io file or a .py file, depending on the languages available to you
4. Observe that instead of the contents of the file you requested, Studio displays the default file (and not even your changed version of it)

[What did you expect to happen?  What happened?  All the time?]
This happens consistently.

[Are there any error messages printed to the console?]
Nothing is printed to the console.

### Versions

- Operating system: Windows 10 *and* Ubuntu 20
- `libfive` commit hash: 4dc715fdbd7595a7957677af327010e27c942cbc
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.12 on Ubuntu, unsure on Windows

mkeeter, 2021-03-27:
lol, I'll check it out

mkeeter, 2021-03-27:
Okay, should be fixed.  Loading files called `guessLanguage()`, which called `setLanguage()`, which called `loadDefaultScript()`, so I uncoupled the latter two.

closed 2021-03-27 by mkeeter

omgitsraven, 2021-03-27:
Fixed, thank you :)

commit d746f91a (2021-03-27): Don't always load the default script (fixes #411)