libfive/libfive #413: Studio crashes on opening Python files on Ubuntu 20

omgitsraven, 2021-03-27:
### Description

When I open a Python file in Studio on Ubuntu 20, I see its contents appear in the left pane very briefly, and then the window instantly closes. It's fine with Scheme files.

If I launch it from the terminal, nothing is printed there UNTIL the moment it crashes, at which point it adds
```
ModuleNotFoundError: No module named 'libfive'
Segmentation fault (core dumped)
```
before returning me to the prompt.

### Versions

- Operating system: Ubuntu 20.04.2 LTS
- `libfive` commit hash: 16c2a8e4e8a6c8119a0239a888807f915d0a5f40
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.12

mkeeter, 2021-03-27:
Good catch – right now, it doesn't install Python libraries system-wide, so it can't find them.  Let me see if this is an easy fix; otherwise, running from the build directory should still work.

mkeeter, 2021-03-27:
Okay, _maybe_ fixed in `41b949647c7e680df8d94cc684941bd3a7cfa4c2` + `d925f03cdb4a8fc4308d6b81072a86c5ef72e0d1`.

Re-run `sudo make install` and it should copy all Python files to the `site-packages` directory, such that if you run a `python3` shell, `import libfive` will work from anywhere in the filesystem.

omgitsraven, 2021-03-27:
Success! Thank you again :)

closed 2021-03-27 by omgitsraven