libfive/libfive #414: Studio is multiple times slower on Windows on certain files

omgitsraven, 2021-03-27:
At long long last -- the issue that I went to all this trouble to report :)

### Description

[Description of the issue]
The same file takes enormously longer to finish meshing on Windows than it does in Linux, even when Windows is native and Linux is in a VM.

[Steps to reproduce]
Open the following file in Studio:
[DEBUG_slowshape.zip](https://github.com/libfive/libfive/files/6216305/DEBUG_slowshape.zip) (unzip it first, Github won't let me attach a .py file...)
and time how long it takes the spinner to disappear.
Do this on both Ubuntu 20 and on Windows.

[What did you expect to happen?  What happened?  All the time?]
I would've expected native execution to be *at least* as fast as a VM on the same machine :)
This happens consistently with **this** file, but there are other files whose rendering time is considerably *faster* on Windows (as I would expect from Windows not being a VM on my machine)

[Are there any error messages printed to the console?]
If there's a way to view that on Windows, I'm not aware of it (when I enter `.\Studio.exe` in Powershell, it launches Studio but gives me another prompt immediately, as though it isn't waiting for any output...)

### Versions

- Operating system: Windows 10 19042
- `libfive` commit hash: d925f03cdb4a8fc4308d6b81072a86c5ef72e0d1
  - Is this the latest commit to `master`?  yes

### Additional Information

On my machine, the file takes 3 seconds to render on virtualized Ubuntu, and 13 seconds to render on native Windows.

The problem seems to be exponential; this is the most stripped-down I could make the file while still reproducing the bug, but there's a much more verbose version that takes 6 seconds on virtualized Ubuntu and 50 seconds on native Windows!

On the other hand, a different file takes 6 seconds on virtualized Ubuntu and 3 seconds on native Windows, which is more what I'd expect given that I only give the VM half my cores. (And yes, I was always shutting down the VM before doing any tests on Windows.)

mkeeter, 2021-03-27:
Very weird!

One easy thing to check is whether the shape is producing the same optimized f-rep, e.g. replacing the last clause with
```
str(cylinder_round_y(cylR,cylL).\
    call(blend,
        sphere(cylR).\
      call(blend,
                sphere(cylR),
            0.12),
        0.15
    ).optimized())
```
(then looking in the output box)

For reference, on macOS I see
```
'(- (* (log (+ (exp (- (* (min (max (- y (+ (* z 4.37114e-08) 0.355)) (- (* z 4.37114e-08) y) (- (sqrt (+ (square (- (+ (* y 4.37114e-08) z))) (square x))) 0.09)) (- (sqrt (+ (square (- y 0.355)) (square x) (square z))) 0.09)) 122.222))) (exp (* (log (* (exp (- 17.1875 (* (sqrt (+ (square x) (square y) (square z))) 190.972))) 2)) 0.64)))) 0.00818182))'
```

I'm also not sure whether Windows is building with optimizations, so try adding `/O2` [here](https://github.com/libfive/libfive/blob/master/CMakeLists.txt#L63) and see if that makes a difference (you should see it added as an option in the build log, I think).

omgitsraven, 2021-03-27:
On Windows:
```
'(- (* (log (+ (exp (* (log (* (exp (- 17.1875 (* (sqrt (+ (square x) (square z) (square y))) 190.972))) 2)) 0.64)) (exp (- (* (min (max (- (sqrt (+ (square (- (+ (* y 4.37114e-08) z))) (square x))) 0.09) (- y (+ (* z 4.37114e-08) 0.355)) (- (* z 4.37114e-08) y)) (- (sqrt (+ (square (- y 0.355)) (square x) (square z))) 0.09)) 122.222))))) 0.00818182))'
```

On Ubuntu:
```
'(- (* (log (+ (exp (* (log (* (exp (- 17.1875 (* (sqrt (+ (square y) (square x) (square z))) 190.972))) 2)) 0.64)) (exp (- (* (min (- (sqrt (+ (square (- y 0.355)) (square x) (square z))) 0.09) (max (- y (+ (* z 4.37114e-08) 0.355)) (- (sqrt (+ (square x) (square (- (+ (* y 4.37114e-08) z))))) 0.09) (- (* z 4.37114e-08) y))) 122.222))))) 0.00818182))'
```

It looks like all three are different??

omgitsraven, 2021-03-27:
Also I'm still not really familiar with compiler stuff; by "adding `/O2`" do you mean changing
```
set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} /MD")
```
to
```
set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} /MD /O2")
```
?

mkeeter, 2021-03-27:
Yes, that's what I meant – that set of flags is compiler flags for MSVC, and `/O2` is "optimize for speed", equivalent to `-O3` on Clang / GCC.  I'll take a look at the different math strings and see if anything is suspicious...

mkeeter, 2021-03-27:
Oh, you could also try adding `EIGEN_NO_DEBUG` to that line, so it looks like

```
    set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} /MD /O2 /DEIGEN_NO_DEBUG")
```

omgitsraven, 2021-03-28:
No change after recompiling with the modified line :(

mkeeter, 2021-03-28:
Can you try running the mesh performance breakdown on each OS?

On Linux, it would be something like
```
./libfive/test/libfive-test "Mesh::render (gyroid performance breakdown)"
```

omgitsraven, 2021-03-28:
There doesn't seem to be a libfive-test file at that location?
```
hazel@hazel-VB20:~/stuff/libfive/libfive/libfive/test$ ls
api.cpp         eval_array.cpp        heightmap.cpp        neighbors.cpp       simplex.cpp             voxels.cpp
archive.cpp     eval_deriv_array.cpp  hybrid_meshing.cpp   object_pool.cpp     solver.cpp              xtree.cpp
catch.hpp       eval_feature.cpp      indexes.cpp          oracle_context.cpp  surface_edge_map.cpp
CMakeLists.txt  eval_interval.cpp     main.cpp             oracle.cpp          transformed_oracle.cpp
contours.cpp    eval_jacobian.cpp     manifold_tables.cpp  progress.cpp        tree.cpp
deck.cpp        feature.cpp           marching.cpp         qef.cpp             util
dual.cpp        guile.cpp             mesh.cpp             region.cpp          vol_tree.cpp
hazel@hazel-VB20:~/stuff/libfive/libfive/libfive/test$ 
```

mkeeter, 2021-03-28:
Oh, you'll want to adjust the build command to build everything, rather than just Studio, e.g. 
```
"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build . --config Release  -- -v:n -m:8
```
(untested)

Then, the `libfive-test` executable will _probably_ be in `build\libfive\test\Release\libfive-test.exe`, or something like that.

omgitsraven, 2021-03-28:
What would the command be for Linux?

mkeeter, 2021-03-28:
On Linux, it's in `build/libfive/test/libfive-test` (note that it's in the build directory, not the source directory).

omgitsraven, 2021-03-28:
Running libfive-test.exe from the command line has no result (not even an error), but double-clicking it produces this:
[image]

mkeeter, 2021-03-28:
Oh no, Windows...

I've added a post-build operation to copy the DLLs into the `build/libfive/test` directory, so `libfive-test.exe` should be able to find them (8b9afa52ef5f7c3837ceb8da9456b03029b4f369).

omgitsraven, 2021-03-28:
oof
[image]

mkeeter, 2021-03-28:
Does this still happen when running on the command line?  I'd suggest continuing to copy DLLs from _somewhere_ in the `vcpkg` directory into `build/libfive/test` directory, until `libfive-test.exe` stops complaining.

(but it's weird that Studio doesn't complain about this, since it should be linked against the same libraries?)

omgitsraven, 2021-03-28:
Command line just gave no reaction, like before. Anyway, eventually I copied in enough DLLs that it worked.

Here it is on Windows:
```
C:\OtherPrograms\libfive\libfive\build\libfive\test\Release>libfive-test.exe "Mesh::render (gyroid performance breakdown)"

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test.exe is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Mesh::render (gyroid performance breakdown)
-------------------------------------------------------------------------------
C:\OtherPrograms\libfive\libfive\libfive\test\mesh.cpp(230)
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
DCTree construction                                 1   1895358900    1.89536 s
Mesh building                                       1     48303000    48.303 ms
DCTree deletion                                     1     96710400   96.7104 ms
Mesh deletion                                   10000        18400         1 ns

===============================================================================
test cases: 1 | 1 passed
assertions: - none -
```

Here it is on Linux:
```
hazel@hazel-VB20:~/stuff/libfive/libfive/build/libfive/test$ ./libfive-test "Mesh::render (gyroid performance breakdown)"

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Mesh::render (gyroid performance breakdown)
-------------------------------------------------------------------------------
/home/hazel/stuff/libfive/libfive/libfive/test/mesh.cpp:230
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
DCTree construction                                 1   2601781239    2.60178 s 
Mesh building                                       1    349177650   349.178 ms 
DCTree deletion                                     1     36837986    36.838 ms 
Mesh deletion                                       1       679397   679.397 µs 

===============================================================================
test cases: 1 | 1 passed
assertions: - none -
```

mkeeter, 2021-03-28:
Okay, that's showing Windows being faster than Linux VM, which is expected.  I notice that the shape that's showing weirdly bad performance is using `log` / `exp`, so I wonder if they're slower on Windows for some reason?

A second theory is that Windows vs Linux are vectorizing differently.  You could test this by sprinkling different `#error` messages [here](https://github.com/libfive/libfive/blob/master/libfive/src/eval/eval_array.cpp#L80-L92), e.g. something like

```
#if defined EIGEN_VECTORIZE_AVX512
    #define LIBFIVE_SIMD_SIZE 16
    #error "Vectorizing to AVX512"
#elif defined EIGEN_VECTORIZE_AVX
    #define LIBFIVE_SIMD_SIZE 8
    #error "Vectorizing to AVX"
#elif defined EIGEN_VECTORIZE_SSE
    #define LIBFIVE_SIMD_SIZE 4
    #error "Vectorizing to SSE"
#elif defined EIGEN_VECTORIZE
    #warning "EIGEN_VECTORIZE is set but no vectorization flag is found"
    #define LIBFIVE_SIMD_SIZE 0
#else
    #warning "No SIMD flags detected"
    #define LIBFIVE_SIMD_SIZE 0
#endif
```
then recompile and see error message is printed on each OS.

omgitsraven, 2021-03-28:
On Windows:
```
C:\OtherPrograms\libfive\libfive\build>"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build . --config Release --target Studio -- -v:n -m:8
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2021-03-28 2:34:19 PM.
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" on node 1 (default targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
     2>InitializeBuildStatus:
         Creating "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         All outputs are up-to-date.
       FinalizeBuildStatus:
         Deleting file "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
         Touching "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
     2>Done Building Project "C:\OtherPrograms\libfive\libfive\build\ZERO_CHECK.vcxproj" (default targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\libfive\bind\python\libfive-python.vcxproj" (3) on node 2 (default targets).
     3>InitializeBuildStatus:
         Creating "x64\Release\libfive-python\libfive-python.tlog\unsuccessfulbuild" because "AlwaysCreate" was specifi
         ed.
       FinalizeBuildStatus:
         Deleting file "x64\Release\libfive-python\libfive-python.tlog\unsuccessfulbuild".
         Touching "x64\Release\libfive-python\libfive-python.tlog\libfive-python.lastbuildstate".
     3>Done Building Project "C:\OtherPrograms\libfive\libfive\build\libfive\bind\python\libfive-python.vcxproj" (defau
       lt targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\libfive\src\libfive.vcxproj" (5) on node 2 (default targets).
     5>InitializeBuildStatus:
         Creating "libfive.dir\Release\libfive.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         All outputs are up-to-date.
       ClCompile:
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\CL.exe
          /c /IC:\OtherPrograms\libfive\libfive\libfive\src\..\include /I"C:\OtherPrograms\libfive\libfive\vcpkg\instal
         led\x64-windows\include" /nologo /W1 /WX /diagnostics:column /O2 /Ob2 /D _USE_MATH_DEFINES /D _SCL_SECURE_NO_W
         ARNINGS /D NDEBUG /D "CMAKE_INTDIR=\"Release\"" /D libfive_EXPORTS /D _WINDLL /D _MBCS /Gm- /EHsc /MD /GS /fp:
         precise /Zc:wchar_t /Zc:forScope /Zc:inline /std:c++17 /Fo"libfive.dir\Release\\" /Fd"libfive.dir\Release\vc14
         2.pdb" /Gd /TP /wd4267 /wd4244 /wd4305 /errorReport:queue  /bigobj C:\OtherPrograms\libfive\libfive\libfive\sr
         c\eval\eval_array.cpp
         eval_array.cpp
     5>C:\OtherPrograms\libfive\libfive\libfive\src\eval\eval_array.cpp(88,1): fatal error C1189: #error:  "Vectorizing
        to SSE" [C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj]
     5>Done Building Project "C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj" (default targets) --
       FAILED.
     1>Done Building Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (default targets) -- FAILED
       .

Build FAILED.

       "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (default target) (1) ->
       "C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj" (default target) (5) ->
       (ClCompile target) ->
         C:\OtherPrograms\libfive\libfive\libfive\src\eval\eval_array.cpp(88,1): fatal error C1189: #error:  "Vectorizi
       ng to SSE" [C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:04.28
```

on Linux:
```
-- Checking for module 'guile-2.2>=2.2.1'
--   No package 'guile-2.2' found
-- Checking for module 'guile-3.0>=3.0.0'
--   Found guile-3.0, version 3.0.1
[... 10 lines cut ...]
[  7%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/deck.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_jacobian.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_interval.cpp.o
[  9%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_array.cpp.o
/home/hazel/stuff/libfive/libfive/libfive/src/eval/eval_array.cpp:85:6: error: #error "Vectorizing to AVX"
   85 |     #error "Vectorizing to AVX"
      |      ^~~~~
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_deriv_array.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/eval_feature.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/tape.cpp.o
make[2]: *** [libfive/src/CMakeFiles/libfive.dir/build.make:115: libfive/src/CMakeFiles/libfive.dir/eval/eval_array.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:218: libfive/src/CMakeFiles/libfive.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
[... cut ...]
/home/hazel/stuff/libfive/libfive/libfive/src/eval/eval_array.cpp:85:6: error: #error "Vectorizing to AVX"
   85 |     #error "Vectorizing to AVX"
      |      ^~~~~
make[2]: *** [libfive/src/CMakeFiles/libfive.dir/build.make:115: libfive/src/CMakeFiles/libfive.dir/eval/eval_array.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:218: libfive/src/CMakeFiles/libfive.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

I hope this isn't related to my having mucked around with compiling Eigen on Windows in the past, for a different project (that I never even got working)...

mkeeter, 2021-03-28:
Okay, looks like Linux is using AVX (8-wide), while Windows is using SSE (4-wide) float instructions.  There are AVX/SSE versions of `exp` and `log` ([source](https://eigen.tuxfamily.org/dox/group__CoeffwiseMathFunctions.html)), so this makes me suspicious!

Can you try adding `/arch:AVX` or `/arch:AVX2` to the MSVC flags (in the [same place](https://github.com/libfive/libfive/blob/master/CMakeLists.txt#L63) as before)?

There are more flags to try in this similar issue [here](https://github.com/cuberite/cuberite/issues/4784#issuecomment-664676589), if that doesn't help.

omgitsraven, 2021-03-28:
Tried `/arch:AVX2`; no speed difference.
Tried `/Ob2 /Oi /Ot /GF /Gm- /fp:fast /arch:AVX /Qpar` and it got SLOWER (18 seconds rather than 13)

mkeeter, 2021-03-28:
With `/arch:AVX` and all of the `#error` hints (to see which path is actually compiled), is it using the `EIGEN_VECTORIZE_AVX` path?

omgitsraven, 2021-03-28:
`set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE} /MD /Ob2 /Oi /Ot /GF /Gm- /fp:fast /arch:AVX /Qpar")` gives
```
C:\OtherPrograms\libfive\libfive\build>"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build . --config Release --target Studio -- -v:n -m:8
CMake is re-running because C:/OtherPrograms/libfive/libfive/build/CMakeFiles/generate.stamp is out-of-date.
  the file 'C:/OtherPrograms/libfive/libfive/CMakeLists.txt'
  is newer than 'C:/OtherPrograms/libfive/libfive/build/CMakeFiles/generate.stamp.depend'
  result='-1'
-- Selecting Windows SDK version 10.0.18362.0 to target Windows 10.0.19042.
CMake Warning (dev) at C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (Boost) does
  not match the name of the calling package (boost).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  C:/Program Files (x86)/Microsoft Visual Studio/2019/Community/Common7/IDE/CommonExtensions/Microsoft/CMake/CMake/share/cmake-3.17/Modules/FindBoost.cmake:2145 (find_package_handle_standard_args)
  vcpkg/scripts/buildsystems/vcpkg.cmake:861 (_find_package)
  CMakeLists.txt:96 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

Checking dependencies:
  libfive:              ✓
  Guile bindings:       ✘   (needs Guile 2.2 or later)
  Python bindings:      ✓
  Studio:               ✓   (Python only)
-- Configuring done
-- Generating done
-- Build files have been written to: C:/OtherPrograms/libfive/libfive/build
Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Framework
Copyright (C) Microsoft Corporation. All rights reserved.

Build started 2021-03-28 3:53:50 PM.
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" on node 1 (default targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\ZERO_CHECK.vcxproj" (2) on node 1 (default targets).
     2>InitializeBuildStatus:
         Creating "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         All outputs are up-to-date.
       FinalizeBuildStatus:
         Deleting file "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\unsuccessfulbuild".
         Touching "x64\Release\ZERO_CHECK\ZERO_CHECK.tlog\ZERO_CHECK.lastbuildstate".
     2>Done Building Project "C:\OtherPrograms\libfive\libfive\build\ZERO_CHECK.vcxproj" (default targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\libfive\bind\python\libfive-python.vcxproj" (3) on node 2 (default targets).
     3>InitializeBuildStatus:
         Creating "x64\Release\libfive-python\libfive-python.tlog\unsuccessfulbuild" because "AlwaysCreate" was specifi
         ed.
       FinalizeBuildStatus:
         Deleting file "x64\Release\libfive-python\libfive-python.tlog\unsuccessfulbuild".
         Touching "x64\Release\libfive-python\libfive-python.tlog\libfive-python.lastbuildstate".
     3>Done Building Project "C:\OtherPrograms\libfive\libfive\build\libfive\bind\python\libfive-python.vcxproj" (defau
       lt targets).
     1>Project "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (1) is building "C:\OtherPrograms\libfive
       \libfive\build\libfive\src\libfive.vcxproj" (5) on node 2 (default targets).
     5>InitializeBuildStatus:
         Creating "libfive.dir\Release\libfive.tlog\unsuccessfulbuild" because "AlwaysCreate" was specified.
       CustomBuild:
         All outputs are up-to-date.
       ClCompile:
         C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.26.28801\bin\HostX64\x64\CL.exe
          /c /IC:\OtherPrograms\libfive\libfive\libfive\src\..\include /I"C:\OtherPrograms\libfive\libfive\vcpkg\instal
         led\x64-windows\include" /nologo /W1 /WX /diagnostics:column /O2 /Ob2 /Oi /Ot /D _USE_MATH_DEFINES /D _SCL_SEC
         URE_NO_WARNINGS /D NDEBUG /D "CMAKE_INTDIR=\"Release\"" /D libfive_EXPORTS /D _WINDLL /D _MBCS /GF /Gm- /EHsc
         /MD /GS /Qpar /arch:AVX /fp:fast /Zc:wchar_t /Zc:forScope /Zc:inline /std:c++17 /Fo"libfive.dir\Release\\" /Fd
         "libfive.dir\Release\vc142.pdb" /Gd /TP /wd4267 /wd4244 /wd4305 /errorReport:queue  /bigobj C:\OtherPrograms\l
         ibfive\libfive\libfive\src\eval\eval_array.cpp
         eval_array.cpp
     5>C:\OtherPrograms\libfive\libfive\libfive\src\eval\eval_array.cpp(85,1): fatal error C1189: #error:  "Vectorizing
        to AVX" [C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj]
     5>Done Building Project "C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj" (default targets) --
[... 204 lines cut ...]
         operations.cpp
         oracle_clause.cpp
         transformed_oracle.cpp
         transformed_oracle_clause.cpp
         Generating Code...
     4>Done Building Project "C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj" (default targets) --
       FAILED.
[... cut ...]
       "C:\OtherPrograms\libfive\libfive\build\studio\Studio.vcxproj" (default target) (1) ->
       "C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj" (default target) (4) ->
       (ClCompile target) ->
         C:\OtherPrograms\libfive\libfive\libfive\src\eval\eval_array.cpp(85,1): fatal error C1189: #error:  "Vectorizi
       ng to AVX" [C:\OtherPrograms\libfive\libfive\build\libfive\src\libfive.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:04:45.32
```

Bizarre that both claim to be using AVX, despite not improving performance over the normal build??

mkeeter, 2021-04-02:
This remains mysterious!  I added more reasonable default flags in 41ed3c2b578a6c8b6b42771c3778a3e2860e066a, but I'm pretty sure that's a combination that you've already tried.

I'm not sure when I'll have time to debug this further, but the next steps are probably
- Extract the troublesome shape into a C++ test case (e.g. [this](https://github.com/libfive/libfive/blob/master/libfive/test/mesh.cpp#L193)), and see if it's still slow
- If not, the issue is in Python (this is unlikely)
- If it's still slow in C++, then profile both machines and see what's function is slow in Windows and not Linux