libfive/libfive #415: Anyone else running Studio on an M1 Mac?

ornamentist, 2021-04-04:
Is anyone else running Studio on an M1 Mac?  It's crashing on startup for me, but probably hasn't been built as a universal binary yet?

mkeeter, 2021-04-05:
I'm on an x86 Macbook Pro, so can confirm that the app bundle on the website isn't universal.  You'll have to try building from source!

Feel free to keep this issue open and let me know how it goes – I'm most concerned about Guile, so you may have to do a Python-only build.

ornamentist, 2021-04-06:
Thanks -- I'll explore building from source further.

ornamentist, 2021-04-06:
Okay, after doing:

```
brew install cmake pkg-config eigen libpng boost guile python3 qt5
```

Then:

```
# clone into lib five directory
cd libfive
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=/opt/homebrew/Cellar/qt@5/5.15.2/lib/cmake/Qt5Core/  ..
# Note the changed Homebrew directories for M1/Big Sur
make
```

I see:

```
[  1%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o
clang: error: the clang compiler does not support '-march=native'
```

Which looks to be related to https://stackoverflow.com/questions/65966969/why-does-march-native-not-work-on-apple-m1 and https://github.com/DMOJ/judge-server/issues/303

mkeeter, 2021-04-06:
Well, that one's easy: can you remove `-march=native` [here](https://github.com/libfive/libfive/blob/master/CMakeLists.txt#L50), and see what breaks next?

ornamentist, 2021-04-06:
Okay, I removed that flag and now seeing:

```
[  1%] Building CXX object libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o
In file included from /Users/stu/tmp/libfive/libfive/src/eval/base.cpp:13:
In file included from /Users/stu/tmp/libfive/libfive/src/../include/libfive/eval/deck.hpp:15:
In file included from /Users/stu/tmp/libfive/libfive/src/../include/libfive/oracle/oracle.hpp:16:
In file included from /Users/stu/tmp/libfive/libfive/src/../include/libfive/eval/interval.hpp:20:
In file included from /opt/homebrew/include/boost/numeric/interval.hpp:18:
/opt/homebrew/include/boost/numeric/interval/hw_rounding.hpp:42:4: error:
      Boost.Numeric.Interval: Please specify rounding control mechanism.
#  error Boost.Numeric.Interval: Please specify rounding control mechanism.
   ^
1 error generated.
make[2]: *** [libfive/src/CMakeFiles/libfive.dir/eval/base.cpp.o] Error 1
make[1]: *** [libfive/src/CMakeFiles/libfive.dir/all] Error 2
make: *** [all] Error 2
```

Something to do with the native architecture floating point macros?

mkeeter, 2021-04-06:
Coincidentally, it looks like this was fixed in Boost [just a few days ago](https://github.com/boostorg/interval/commit/1b95966ad06afd57f4a124902f81398232503624).  I'm not sure whether installing from `HEAD` in Homebrew will pull this, but the patch is there if you want to apply it to your local Boost installing.

ornamentist, 2021-04-07:
Okay, tried installing boost from HEAD and saw this:

```
==> Patching
==> Applying 7b3fc734242eea9af734d6cd8ccb3d8f6b64c5b2.patch
patching file boost-install.jam
Hunk #1 FAILED at 483.
1 out of 1 hunk FAILED -- saving rejects to file boost-install.jam.rej

Do not report this issue to Homebrew/brew or Homebrew/core!

These open issues may also help:
boost*: move from bintray https://github.com/Homebrew/homebrew-core/pull/74338
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.
```

Will try manually applying the patch to the non-HEAD version too.

ornamentist, 2021-04-07:
I reinstalled the non-HEAD version of Boost, manually applied that patch and got a lot further before seeing this error:

```
[ 46%] Building CXX object libfive/test/CMakeFiles/libfive-test.dir/main.cpp.o
In file included from /Users/stu/tmp/libfive/libfive/test/main.cpp:11:
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:5060:13: error: invalid token in expression
            CATCH_BREAK_INTO_DEBUGGER();
            ^
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:4782:75: note: expanded from macro
      'CATCH_BREAK_INTO_DEBUGGER'
    #define CATCH_BREAK_INTO_DEBUGGER() if( Catch::isDebuggerActive() ) { CATCH_TRAP(); }
                                                                          ^
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:4761:34: note: expanded from macro 'CATCH_TRAP'
    #define CATCH_TRAP() __asm__("int $3\n" : : ) /* NOLINT */
                                 ^
<inline asm>:1:6: note: instantiated into assembly here
        int $3
            ^
In file included from /Users/stu/tmp/libfive/libfive/test/main.cpp:11:
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:5060:13: error: invalid operand
            CATCH_BREAK_INTO_DEBUGGER();
            ^
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:4782:75: note: expanded from macro
      'CATCH_BREAK_INTO_DEBUGGER'
    #define CATCH_BREAK_INTO_DEBUGGER() if( Catch::isDebuggerActive() ) { CATCH_TRAP(); }
                                                                          ^
/Users/stu/tmp/libfive/libfive/test/./catch.hpp:4761:34: note: expanded from macro 'CATCH_TRAP'
    #define CATCH_TRAP() __asm__("int $3\n" : : ) /* NOLINT */
                                 ^
<inline asm>:1:6: note: instantiated into assembly here
        int $3
            ^
2 errors generated.
make[2]: *** [libfive/test/CMakeFiles/libfive-test.dir/main.cpp.o] Error 1
make[1]: *** [libfive/test/CMakeFiles/libfive-test.dir/all] Error 2
```

mkeeter, 2021-04-07:
That's probably fixed by updating [Catch](https://github.com/catchorg/Catch2), which is vendored into the repo [here](https://github.com/libfive/libfive/blob/master/libfive/test/catch.hpp).  I'll put this on my list, but am very busy at work and lacking in spare time right now.

Catch is only used by `libfive-test`, so you can try just compiling `Studio` (the GUI).  This is `ninja studio/all` if you're using [Ninja](https://ninja-build.org/), and probably something similar with Make.

ornamentist, 2021-04-07:
No worries and thanks so much for the work you put into libfive.

andreypopp, 2021-04-10:
Based on what was found in this issue I've created a PR which updates Catch2 and finally builds the libfive on Apple Silicon. Thanks for the investigation here!

szhorvat, 2021-06-02:
FYI, here's the log from the failed build on Apple Silicon on the MacPorts buildbot: https://build.macports.org/builders/ports-11_arm64-builder/builds/20751/steps/install-port/logs/stdio

ornamentist, 2021-06-03:
Have the upstream packages needed to compile changed?  When I run cmake it's complaining about Qt5 widgets:

```
By not providing "FindQt5Widgets.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Qt5Widgets", but CMake did not find one.
```

I don't know much about Qt so I don't see any Homebrew packages or casks that seem to refer to that.

mkeeter, 2022-03-16:
This should now be working, see #420 !

closed 2022-03-16 by mkeeter

tecosaur, 2025-08-17:
I don't suppose a pre-made `.dmg` installer for aarch64 mac could be made available?