libfive/libfive #417: `for` loop in Python produces no result

omgitsraven, 2021-04-08:
[Description of the issue]
Trying to use a `for` loop just prints a result of `None` and produces no geometry.

[Steps to reproduce]
Open this file:
```
import studio
studio.set_bounds([-10, -10, -10], [10, 10, 10])
studio.set_quality(8)
studio.set_resolution(10)

from libfive.stdlib import *

for n in range(3):
    sphere(1)
```

### Versions

- Operating system: Windows 10
- `libfive` commit hash: 533045072d679cba7d400f1b2a97a2cc31efb91e
  - Is this the latest commit to `master`?  yes

mkeeter, 2021-04-09:
I believe this is behaving correctly; the Scheme equivalent is
```
(do ((i 0 (1+ i)))
    ((> i 2))
    (sphere 1))
```
which also doesn't show a shape.

In general, the script interpreter runs each **expression**, and if the result of that expression is a `Shape`, then it renders that shape in the viewport.

`sphere(1)` is an expression that evaluates to a `Shape`.

Your example of
```
for n in range(3):
    sphere(1)
```
is a (single) expression that runs a loop and evaluates to `None`.

Just because a shape was created at some point, doesn't mean it will be rendered in the UI:
a simple example is `sphere(1).difference(cylinder(0.5))`, which constructs a sphere and cylinder shape while running, but only draw the resulting CSG primitive.

omgitsraven, 2021-04-09:
Ahh! I actually hadn't thought about it that way, but I guess that makes sense.

For anyone else digging up this thread, based on that information, I wound up doing this instead:

```
myshape = sphere(1)
for n in range(3):
    myshape = myshape.union(sphere(1).move([n,0,0]))
myshape
```
which works! Thanks!

closed 2021-04-09 by omgitsraven