libfive/libfive #418: `emptiness` no longer exists in either language

omgitsraven, 2021-04-08:
I went to try and make a recursive `for` loop in Python, and needed an empty shape to pass it to begin with, and found that `emptiness` didn't seem to be declared.
But then I went and tried the same thing in Scheme to be sure that I had remembered the name correctly, and it was undefined there too...
But when I search it in the repository, it seems like I was remembering the name correctly? https://github.com/libfive/libfive/commit/92b298613d2e50e64c98dbeb39b5deb14910f991

I'm assuming this is a case of it not being ported to the new C stdlib?

mkeeter, 2021-04-09:
Yes, I hadn't ported it because the C library didn't support zero-argument functions.  It should be fixed now!

Note that it's now returning `+INFINITY` instead of 1, which means doing `min(..., emptiness())` won't clamp the distance field to 1; this kind of clamping is bad for certain use cases, but let me know if anything breaks!

closed 2021-04-09 by mkeeter

omgitsraven, 2021-04-09:
I hadn't even realised that it wasn't Infinity until now, so my stuff should be fine at least! Thanks :)

commit 289b920e (2021-04-09): Add emptiness and support for no-argument functions

Fixes #418