libfive/libfive #419: How to generate a new shape in Python?

omgitsraven, 2021-04-09:
I've tried digging through the source in https://github.com/libfive/libfive/blob/7824deacfa73830d08970a52511f35743e441c64/libfive/bind/python/libfive/shape.py but I confess that my python isn't good enough to know how I'm meant to invoke that `shape` function at the end... `shape()` isn't defined, `Shape.shape()` isn't defined, and if I write `Shape(lambda x,y,z: sqrt(x*x+y*y+z*z)-1)` then Studio immediately crashes...

mkeeter, 2021-04-09:
`shape` is a [decorator](https://realpython.com/primer-on-python-decorators/)!

You can apply it to what looks like a function, and it will convert it into a `Shape`.

Here's an example of how to use it:
```python
from libfive.shape import shape

@shape
def my_new_shape(x, y, z):
    return x**2 + y**2 + z**2 - 1

# at this point, my_new_shape is a Shape variable
my_new_shape # to render the sphere in the viewport
```

omgitsraven, 2021-04-09:
Aha -- the part I really needed was `from libfive.shape import shape`! Is there a reason it isn't included in the stdlib import?

Anyway, those all work -- I think I actually prefer the Shape.X() syntax in the end, since that way my custom shape function can actually take other arguments (like a radius)...

One more thing for some eventual examples!

closed 2021-04-09 by omgitsraven