libfive/libfive #424: Multiple test suite failures

szhorvat, 2021-04-24:
### Description

[Description of the issue]

Running `libfive-test` results in this:

```
$ ./libfive-test

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
DerivArrayEvaluator::getAmbiguousDerivs
-------------------------------------------------------------------------------
../libfive/test/eval_deriv_array.cpp:94
...............................................................................

../libfive/test/eval_deriv_array.cpp:106: FAILED:
  REQUIRE( a.count() == 2 )
with expansion:
  3 == 2

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
../libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1     59690058   59.6901 ms
Menger sponge                                       1    162406479   162.406 ms

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
../libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
SOS root                                            1       395083   395.083 µs

../libfive/test/solver.cpp:116: FAILED:
  REQUIRE( res < 1e-3 )
with expansion:
  0.00165f < 0.001

guile: uncaught throw to misc-error: (#f ~A ~S (no code for module (libfive kernel)) #f)
```

[Steps to reproduce]

Build from sources using default options.

```
cd libfive/test
./libfive.test
```

[What did you expect to happen?  What happened?  All the time?]
[Are there any error messages printed to the console?]

Not during build.

### Versions

- Operating system: macOS 10.14.6
- `libfive` commit hash: e7486b9b63c1f67509be02399d09b9411ea6c1c2
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.15.2

### Additional Information

Any additional information, configuration or data that might be necessary to reproduce the issue.

CMake configuration output:

```
-- The C compiler identification is AppleClang 11.0.0.11000033
-- The CXX compiler identification is AppleClang 11.0.0.11000033
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Boost: /opt/local/include (found version "1.71.0")
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.29.2")
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.7
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.9
-- Found ZLIB: /opt/local/lib/libz.dylib (found version "1.2.11")
-- Found PNG: /opt/local/lib/libpng.dylib (found version "1.6.37")
-- Found Python3: /opt/local/bin/python3.9 (found version "3.9.4") found components: Interpreter Development Development.Module Development.Embed
Checking dependencies:
  libfive:		✓
  Guile bindings:	✓
  Python bindings:	✓
  Studio:		✓   (Python + Guile)
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/szhorvat/Repos/libfive/build
```

szhorvat, 2021-04-24:
I also tried to build with the latest LLVM Clang (version 11.1.0 from MacPorts). The failures are different now:

<details><summary>`clang 11.1.0`</summary>
```
~/R/l/build (master|✔) [4] $ libfive/test/libfive-test

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
../libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1     58156713   58.1567 ms
Menger sponge                                       1    170228196   170.228 ms

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
../libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
SOS root                                            1       747692   747.692 µs

-------------------------------------------------------------------------------
make-shape
  Invalid opcode
-------------------------------------------------------------------------------
../libfive/test/guile.cpp:66
...............................................................................

../libfive/test/guile.cpp:70: FAILED:
  REQUIRE( boost::algorithm::starts_with(result, "wrong-type-arg:") )
with expansion:
  false
with message:
  result := "misc-error: Invalid opcode: lol"

-------------------------------------------------------------------------------
make-shape
  Invalid argument count
-------------------------------------------------------------------------------
../libfive/test/guile.cpp:73
...............................................................................

../libfive/test/guile.cpp:77: FAILED:
  REQUIRE( boost::algorithm::starts_with(a, "wrong-number-of-args:") )
with expansion:
  false
with message:
  a := "misc-error: Opcode var-x takes 0 arguments"

-------------------------------------------------------------------------------
make-shape
  Invalid argument type
-------------------------------------------------------------------------------
../libfive/test/guile.cpp:84
...............................................................................

../libfive/test/guile.cpp:88: FAILED:
  REQUIRE( boost::algorithm::starts_with(a, "wrong-type-arg:") )
with expansion:
  false
with message:
  a := "misc-error: Opcode sin takes 1 shape argument"

===============================================================================
test cases:    183 |    182 passed | 1 failed
assertions: 909931 | 909928 passed | 3 failed
```
</details>

mkeeter, 2021-04-24:
Thanks, the Guile errors you saw in your second run are fixed in 63860fed5bd9a2b1f3eff9b29fd81afacfc8eb52.  I'll look into the other tests.

szhorvat, 2021-04-24:
This one runs to completion now, when compiled with Apple Clang. There is an occasional non-deterministic failure:

```
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
DerivArrayEvaluator::getAmbiguousDerivs
-------------------------------------------------------------------------------
../libfive/test/eval_deriv_array.cpp:94
...............................................................................

../libfive/test/eval_deriv_array.cpp:106: FAILED:
  REQUIRE( a.count() == 2 )
with expansion:
  3 == 2

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
../libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1     63850178   63.8502 ms
Menger sponge                                       1    175355644   175.356 ms

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
../libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
SOS root                                            1       758968   758.968 µs

===============================================================================
test cases:    183 |    182 passed | 1 failed
assertions: 910028 | 910027 passed | 1 failed
```

mkeeter, 2021-04-24:
Great, I'll leave this open and see if I can reproduce the non-deterministic failure.